espressif/esp-matter #1129: "Crash in `light_wifi_prov` example during Node Initialization - Failed to get device certificate"

iamiha, 2024-10-24:
**Description:**

I'm encountering a crash while running the `light_wifi_prov` example from ESP-Matter. The crash occurs during the initialization of the Matter node, and the log shows errors related to obtaining the device certificate from flash. Below is the relevant portion of the log and backtrace.

**Environment:**
- ESP-IDF Version: v5.2.3
- Board: ESP32-S3
- ESP-Matter: using the `light_wifi_prov` example

**Logs:**

```
E (828) esp_rmaker_client_data: Failed to obtain flash address of device cert
I (828) esp_rmaker_client_data: Attempting to fetch client certificate from NVS
E (838) esp_rmaker_core: Failed to get device certificate.
E (848) esp_rmaker_core: Failed to initialise Node Id. Please perform "claiming" using RainMaker CLI.
E (858) app_main: Could not initialise node. Aborting!!!
abort() was called at PC 0x4200a979 on core 0
```

**Backtrace:**
```
0x40375b92: panic_abort at components/esp_system/panic.c:466
0x40381679: esp_system_abort at components/esp_system/port/esp_system_chip.c:93
0x403898d6: abort at components/newlib/abort.c:38
0x4200a979: app_main at main/app_main.cpp:237 (discriminator 1)
0x4214879f: main_task at components/freertos/app_startup.c:208
0x4038210a: vPortTaskWrapper at components/freertos/portable/xtensa/port.c:134
```

shubhamdp, 2024-10-25:
> E (848) esp_rmaker_core: Failed to initialise Node Id. Please perform "claiming" using RainMaker CLI.

@iamiha plese perform the host based claiming as described in [additional environment setup](https://github.com/espressif/esp-matter/tree/main/examples/light_wifi_prov#2-additional-environment-setup).

dhrishi, 2024-11-13:
I am closing this now. As mentioned above, performing claiming should help.

closed 2024-11-13 by dhrishi