espressif/esp-matter #1130: Build is broken on release/v1.3

immortaleeb, 2024-10-26:
**Describe the bug**
I'm getting the error below when trying to build the light example for esp32c6. I'm using branch release/v1.3.
Looking at the last build report this also seems to be broken on main: https://github.com/espressif/esp-matter/actions/runs/11513101864/job/32049204050#step:4:1518

Anyone know what the latest working state is on release/v1.3?

```
FAILED: esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/font/lv_font_montserrat_14.c.obj
/Users/enver/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.2.1\" -DLV_CONF_INCLUDE_SIMPLE -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/build/config -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/examples -I/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/demos -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/FreeRTOS-Kernel/include/freertos -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/newlib/platform_include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/config/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/config/include/freertos -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/config/riscv/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/FreeRTOS-Kernel/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/freertos/esp_additions/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_hw_support/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_hw_support/include/soc -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_hw_support/include/soc/esp32c6 -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_hw_support/port/esp32c6/. -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_hw_support/port/esp32c6/private_include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/heap/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/log/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/soc/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/soc/esp32c6 -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/soc/esp32c6/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/hal/platform_port/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/hal/esp32c6/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/hal/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_rom/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_rom/include/esp32c6 -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_rom/esp32c6 -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_common/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_system/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_system/port/soc -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_system/port/include/riscv -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_system/port/include/private -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/riscv/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/include/apps -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/include/apps/sntp -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/lwip/src/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/port/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/port/freertos/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/port/esp32xx/include -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/port/esp32xx/include/arch -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/lwip/port/esp32xx/include/sys -I/Users/enver/Projects/Personal/matter/esp-idf-2/components/esp_timer/include -march=rv32imac_zicsr_zifencei  -fdiagnostics-color=always -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fno-shrink-wrap -fmacro-prefix-map=/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light=. -fmacro-prefix-map=/Users/enver/Projects/Personal/matter/esp-idf-2=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -Wno-error=uninitialized -Wno-error=maybe-uninitialized -Wno-missing-field-initializers -Wno-error=array-bounds -Wno-write-strings -Wno-format-nonliteral -Wno-format-security -std=gnu17 -Wno-old-style-declaration -Os -MD -MT esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/font/lv_font_montserrat_14.c.obj -MF esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/font/lv_font_montserrat_14.c.obj.d -o esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/font/lv_font_montserrat_14.c.obj -c /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c
In file included from /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_kconfig.h:15,
                 from /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_internal.h:34,
                 from /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/../../src/lv_init.h:16,
                 from /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/../../lvgl.h:21,
                 from /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:10:
/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/build/config/sdkconfig.h:1039:41: error: expected identifier or '(' before string constant
 1039 | #define CONFIG_LV_ATTRIBUTE_LARGE_CONST ""
      |                                         ^~
/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_internal.h:1324:42: note: in expansion of macro 'CONFIG_LV_ATTRIBUTE_LARGE_CONST'
 1324 |         #define LV_ATTRIBUTE_LARGE_CONST CONFIG_LV_ATTRIBUTE_LARGE_CONST
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:24:8: note: in expansion of macro 'LV_ATTRIBUTE_LARGE_CONST'
   24 | static LV_ATTRIBUTE_LARGE_CONST const uint8_t glyph_bitmap[] = {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:2155:21: error: 'glyph_bitmap' undeclared here (not in a function)
 2155 |     .glyph_bitmap = glyph_bitmap,
      |                     ^~~~~~~~~~~~
[506/506] stamp obj/esp32.stamp
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/build/log/idf_py_stderr_output_68339 and /Users/enver/Projects/Personal/matter/esp-matter-2/examples/light/build/log/idf_py_stdout_output_68339
```

**Environment**
- ESP-Matter Commit Id: f439ecbc91b81f685072fcd64568951dd394fc6a (release/v1.3)
- ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (v5.2.1)
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file): n/a

- Host Machine OS: Mac OS 14.5
- Host Machine Python version: python 3.11.9
...

immortaleeb, 2024-10-26:
Seems to be related to this issue: https://github.com/lvgl/lvgl/issues/7154

immortaleeb, 2024-10-26:
A workaround has been suggested here: https://github.com/espressif/esp-idf/issues/14782#issuecomment-2437989694

Unfortunately, I'm pretty new to esp-idf and can't figure out how I can tell the build system to downgrade the dependency to 9.2.0. Creating an `idf_component.yml` in the `examples/light` directory and removing the `dependencies.lock` file doesn't seem to work.

Whenever I run `idf.py build` it always picks up lvgl version `9.2.1`. Is there some other way to force the build system to not include the faulty version?

immortaleeb, 2024-10-26:
I had to downgrade espressif/led_strip too:

```
$ idf.py add-dependency "espressif/led_strip==2.5.5"
$ idf.py reconfigure
```

Now the build seems to work

rishi9699, 2024-10-27:
@immortaleeb 
The led_strip error that you are getting is due to a recent change in the led_strip repository.
I have created a [pull request](https://github.com/espressif/esp-bsp/pull/423) for fixing the same.

dhrishi, 2024-10-29:
Hi @immortaleeb With the above fix being merged, can you please confirm if you can build the examples fine now? I am able to successfully build the light example on release/v1.3 branch.