espressif/esp-matter #1137: Build causing error issues with connectedhomeip

farhan-syed, 2024-10-31:
**Describe the bug**

Not sure exactly what the bug is. When I add esp-matter as a component dependency. I then try to build and I get tons of errors seeming to do with connectedhomeip. 

I have my project folder 'sample' inside a esp/ folder. Which also contains esp-idf along with esp-matter.

I tried to use the esp-matter that I installed through the guide but it wasn't working so I then tried to use the add-dependency way and that also didnt work.

**Environment**
- ESP-Matter Commit Id: v1.3.0
- ESP-IDF Commit Id: v5.2.1
- SoC (eg: ESP32 or ESP32-C3): esp32c6
- Device Logs (Please attach the log file): 
[build-error.docx](https://github.com/user-attachments/files/17593185/build-error.docx)

- Host Machine OS: macOS Sequoia
- Host Machine Python version: 3.9.6

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

shripad621git, 2024-11-04:
@farhan-syed , you can refer to the `managed_components_light` example [CMakeLists.txt ](https://github.com/espressif/esp-matter/blob/release/v1.3/examples/managed_component_light/CMakeLists.txt)if you wish to use the esp-matter as a managed_component on release/v1.3 branch as it has the esp-matter component 1.3.0.
I got some similar errors when I commented [this](https://github.com/espressif/esp-matter/blob/f439ecbc91b81f685072fcd64568951dd394fc6a/examples/managed_component_light/CMakeLists.txt#L27).
Can you please ensure that these changes in CMakeLists.txt are encorporated in your example.
In case you want to build without esp-matter as a component dependency using existing esp-matter installed , if you are building the project on top of existing esp-matter example, please build your example keeping the esp-matter example as template so that you can get the CMakeLists.txt changes already present.

shripad621git, 2024-11-10:
@farhan-syed , any updates on the issue.

farhan-syed, 2024-11-12:
@shripad621git Sorry! Been busy. Will take a look at the suggestions and give it a go to see if it resolves the issues. Will post back here with an update. Thanks!

farhan-syed, 2024-11-12:
Sorry! Been busy. Will take a look at the suggestions and give it a go to see if it resolves the issues. Will post back here with an update. Thanks!

dhrishi, 2025-03-03:
@farhan-syed I am closing this now. Please re-open when you try it out and if it fails.

closed 2025-03-03 by dhrishi