espressif/esp-matter #1138: Unable to read the Product Label and Serial Number

MaplestoryAlen, 2024-11-01:
I encountered a strange problem where I was able to read properties such as VendorId and ProductName, but not ProductLabel and Serial Number. However, my chip tool was able to read them, and I don't know where the problem lies
   Last week it was still readable, but strangely now it suddenly becomes unreadable

chip-tool:

[image]
[image]

 code:

 	if(dataProvider.GetProductName(device_info.productName, sizeof(device_info.productName)) == CHIP_NO_ERROR)
	{
	   USER_LOGI(TAG,"device_info.productName :%s",device_info.productName); 
	}
	if(dataProvider.GetHardwareVersionString(device_info.HardwareVersionString, sizeof(device_info.HardwareVersionString)) == CHIP_NO_ERROR)		
	{
	   USER_LOGI(TAG,"device_info.HardwareVersionString :%s",device_info.HardwareVersionString);
	}
	if(dataProvider.GetVendorId(device_info.VendorId) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"VendorId :%d",device_info.VendorId); 
	}
	if(dataProvider.GetProductId(device_info.ProductID) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"ProductID :%d",device_info.ProductID); 
	}
	if(dataProvider.GetSetupDiscriminator(device_info.Discriminator) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"device_info.Discriminator :%d",device_info.Discriminator); 
	}	
	char ProductLabel[64] ={0};
	if(dataProvider.GetProductLabel(ProductLabel,strlen(ProductLabel)) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"ProductLabel :%s",ProductLabel); 
	}
	char ReadSerialNumber[32];
	if(dataProvider.GetSerialNumber(ReadSerialNumber,strlen(ReadSerialNumber)) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"ReadSerialNumber :%s",ReadSerialNumber); 
	}
log 

[image]

sdkconfig 

[sdkconfig.txt](https://github.com/user-attachments/files/17596868/sdkconfig.txt)

shripad621git, 2024-11-04:
@MaplestoryAlen , 
```
char ProductLabel[64] ={0};
if(dataProvider.GetProductLabel(ProductLabel,strlen(ProductLabel)) == CHIP_NO_ERROR) 	
{
   USER_LOGI(TAG,"ProductLabel :%s",ProductLabel); 
}
char ReadSerialNumber[32];
if(dataProvider.GetSerialNumber(ReadSerialNumber,strlen(ReadSerialNumber)) == CHIP_NO_ERROR) 	
{
   USER_LOGI(TAG,"ReadSerialNumber :%s",ReadSerialNumber); 
}

```
I guess the above code snippet should have `device_info.ProductLabel` like `device_info.Discriminator` if you are setting them and then using here to retrieve them in the code.

shripad621git, 2024-11-10:
@MaplestoryAlen , any updates on the issue?

MaplestoryAlen, 2024-11-11:
> ，此问题是否有任何更新？
 
I have given up using Matter's API to read, I will use IDF's NVS to read

closed 2024-11-11 by MaplestoryAlen