espressif/esp-matter #1139: BLE Service Disable Abnormal

emesamwong, 2024-11-01:
**Describe the bug**
A clear and concise description of what the bug is.
Hi, I would like to disable the BLE Service Mode function when the DUT is powered up and Enable the BLE Service Mode when Matter is enabled via LCD display control. 

However, when I try to pair the DUT with the Chip tool. It could not pair successfully.
Here is the configuration:
sdkconfig: CONFIG_CHIP_ENABLE_PAIRING_AUTOSTART=n

BLEManagerImpl.cpp: 
 CHIP_ERROR BLEManagerImpl::_Init():
[image]
[image]

By the Log, it seems the BLE level is completed, However, the Chip-tool log is 'Timeout' at PASEsession.
[image]
[image]

**Environment**
- ESP-Matter Commit Id: b5a233744c1f5b61f786339e2912617c81c1e1a0
- ESP-IDF Commit Id: 3b8741b172dc951e18509698dee938304bcf1523
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Device Logs (Please attach the log file):
  DUT Log: [DUT Log.txt](https://github.com/user-attachments/files/17597803/DUT.Log.txt)
  Chip-tool Log: [Chip-Log.txt](https://github.com/user-attachments/files/17597811/Chip-Log.txt)

- Host Machine OS: Linux
- Host Machine Python version: 3.10.12

- Commissioner app and versions if present: chip-tool
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
If I did not make any change in BLEManagerImpl.cpp and did not set "CONFIG_CHIP_ENABLE_PAIRING_AUTOSTART=n":
DUT powered up -> enabled BLE & commissioning window is open -> the pairing is successfully 

Is any configuration or setting required for "Disable BLE Server"? 

Please let me know if any information is required.

pimpalemahesh, 2024-11-07:
@emesamwong The `CONFIG_CHIP_ENABLE_PAIRING_AUTOSTART` option is used to prevent the commissioning window from opening on boot. You mentioned that this option is disabled, but could you explain how you are still able to commission the device using the chip-tool if the commissioning window is closed?

Also do you want to disable BLE entirely on DUT power-up, or just stop advertising on boot and start advertising once Matter is enabled via LCD display control?

dhrishi, 2025-03-03:
@emesamwong Can you please provide the requested inputs? Please close the issue if resolved

closed 2025-05-15 by dhrishi