espressif/esp-matter #1140: where CHIP_DEVICE_CONFIG_ENABLE_THREAD is defined

Buddhika93n, 2024-11-03:
in esp_matter example ICD_APP #if CHIP_DEVICE_CONFIG_ENABLE_THREAD
ESP_LOGI(TAG, "CHIP_DEVICE_CONFIG_ENABLE_THREAD"); this line is printing so CHIP_DEVICE_CONFIG_ENABLE_THREAD is defined somewhere  cannot find it. please help me

shripad621git, 2024-11-04:
@Buddhika93n , the option `CHIP_DEVICE_CONFIG_ENABLE_THREAD` is defined [here](https://github.com/espressif/connectedhomeip/blob/17b1a38e909e7874593bcb87c31be03a5866f1d4/src/platform/ESP32/CHIPDevicePlatformConfig.h#L51) where the option is mapped to `CONFIG_ENABLE_MATTER_OVER_THREAD` in ESP32.
The thread support is enabled by enabling `CONFIG_ENABLE_MATTER_OVER_THREAD`   through idf.py menuconfig.
The option is defined [here](https://github.com/espressif/connectedhomeip/blob/17b1a38e909e7874593bcb87c31be03a5866f1d4/config/esp32/components/chip/Kconfig#L779).

shripad621git, 2024-11-10:
@Buddhika93n , can you please close the issue if it is addressed.

closed 2024-11-11 by Buddhika93n