espressif/esp-matter #1141: Descriptor cluster is missing feature list

jonsmirl, 2024-11-04:
Implementing devices with multiple endpoints requires "Duplicate Condition" to be satisfied.
[image]

Descriptor cluster is missing support for the TAGLIST feature.

[image]

jonsmirl, 2024-11-11:
decriptor::create() is missing the feature list as the last parameter.  I need to enable the feature list.
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.h#L67
[image]
should look like this:
cluster_t *create(endpoint_t *endpoint, config_t *config, uint8_t flags, uint32_t features);

jadhavrohit924, 2024-11-12:
Okay. You can use the `tagList::add()` API to enable the feature.

jadhavrohit924, 2024-11-12:
We will fix the `create` API.

closed 2024-12-05 by espressif-bot