espressif/esp-matter #1142: Remove all network credentials and last fabric infromation once fabric removal request received from Controller

bilalmalik76, 2024-11-04:
How to remove all saved network credentials and start Bluetooth LE advertising once fabric removal request is received from Matter controller. The device shoyld remove all saved network credentials and only keep application-specific non-volatile data? How to achieve this with esp-matter APIs?

pimpalemahesh, 2024-11-06:
@bilalmalik76 To address your requirements for removing network credentials and re-enabling commissioning, there are two approaches to consider:
1. Direct factory reset upon receiving the `kFabricRemoved` event. However, this method will clear all data from nvs partition and "esp_matter_kvs" namespace after factory reset.
`esp_matter::factory_reset()`

3. You can follow these steps to keep application-specific data irrespective of namespace
- Step 1: Use the `esp_wifi_restore()` API to reset the Wi-Fi stack to its default values
- Step 2:  Reopen the commissioning window and restart BLE advertising to allow re-commissioning.
`chip::Server::GetInstance().GetCommissioningWindowManager().OpenBasicCommissioningWindow();
`
- Step 3: Reboot the device
`esp_restart()`

You can take reference from this code snippet:
```
case chip::DeviceLayer::DeviceEventType::kFabricRemoved:
        {
            ESP_LOGI(TAG, "Fabric removed successfully");
            esp_err_t err = esp_wifi_restore();
            if (err != ESP_OK) {
                ESP_LOGE(TAG, "Failed to restore wifi stack");
            }
            CHIP_ERROR error = chip::Server::GetInstance().GetCommissioningWindowManager().OpenBasicCommissioningWindow();
            if (error != CHIP_NO_ERROR)
            {
                ESP_LOGE(TAG, "Failed to open commissioning window");
            }
            esp_restart();
        break;
        }
```

bilalmalik76, 2024-11-09:
@pimpalemahesh Thank you for sharing, I will try this and let you know.

pimpalemahesh, 2024-11-15:
@bilalmalik76 Could you close the issue if it is resolved?

dhrishi, 2024-11-27:
Closing. Please re-open

closed 2024-11-27 by dhrishi