espressif/esp-matter #1146: It is not possible to commission a device using the pairing_code_thread() API

pavel808, 2024-11-07:
Hi all. I had created a general inquiry about this some weeks ago but I hadn't got a response. 

I am working on a Matter Controller application which runs on an ESP32-S3 on a Thread BR device.

When my application calls the commissioning function `esp_matter::controller::pairing_code_thread()`, the commissioning procedure starts but then fails every time with Error on commissioning step `'ThreadNetworkSetup': 'Error CHIP:0x000000AC`.

I can't seem to understand what is wrong exactly. The Thread network is set up properly, and when I try to commission the same device using chip-tool with its pairing code, through the same Thread Border Router, the device commissions ok. 

I have attached the complete logs from the commission attempt from both my Matter Controller application running on the ESP32-S3, and the Thread end device which i'm trying to commission which is running on a ESP32-C6 device.

I'm not saying that there is a bug on your side. However, I have been trying to solve this for a few weeks now without any joy. I have cleared and re-flashed the devices multiple times and re-configured the Thread network etc. 

I'd appreciate any help or ideas. Thanks in advance. 

[device-output-commission-fail (1).log](https://github.com/user-attachments/files/17661680/device-output-commission-fail.1.log)
[server-commissioner-fail (1).log](https://github.com/user-attachments/files/17661682/server-commissioner-fail.1.log)

wqx6, 2024-11-08:
Looks like the dataset you pass to the API is wrong. Note that the API `esp_matter::controller::pairing_code_thread()` should use a bytes input instead of a octet string.

You can verify that the dataset that end-device receives by adding some log at [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/network-commissioning/network-commissioning.cpp#L980).

pavel808, 2024-11-08:
@wqx6 Isn't the octet string an array of bytes anyways in hex format. This format works when passed to chip-tool prepended with `hex: ` . Can I do the same here?

Or can you please give me an example of the exact format that I should be passing, or how to convert what I have.

Many thanks!

wqx6, 2024-11-08:
```
static int char_to_int(char ch)
{
    if ('A' <= ch && ch <= 'F') {
        return 10 + ch - 'A';
    } else if ('a' <= ch && ch <= 'f') {
        return 10 + ch - 'a';
    } else if ('0' <= ch && ch <= '9') {
        return ch - '0';
    }
    return -1;
}

static bool convert_hex_str_to_bytes(const char *hex_str, uint8_t *bytes, uint8_t &bytes_len)
{
    if (!hex_str) {
        return false;
    }
    size_t hex_str_len = strlen(hex_str);
    if (hex_str_len == 0 || hex_str_len % 2 != 0 || hex_str_len / 2 > bytes_len) {
        return false;
    }
    bytes_len = hex_str_len / 2;
    for (size_t i = 0; i < bytes_len; ++i) {
        int byte_h = char_to_int(hex_str[2 * i]);
        int byte_l = char_to_int(hex_str[2 * i + 1]);
        if (byte_h < 0 || byte_l < 0) {
            return false;
        }
        bytes[i] = (byte_h << 4) + byte_l;
    }
    return true;
}

const char *dataset_tlvs = "0e080000000000010000000300001635060004001fffe00208dead00beef00cafe0708fd000db800a00000051000112233445566778899aabbccddaaff030e4f70656e5468726561642d455350010212340410104810e2315100afd6bc9215a6bfac530c0402a0f7f8";

otOperationalDatasetTlvs dataset;
convert_hex_str_to_bytes(dataset_tlvs, dataset.mTlvs, dataset.mLength);
esp_matter::controller::pairing_code_thread(node_id, payload, dataset.mTlvs, dataset.mLength);
```

pavel808, 2024-11-08:
@wqx6 Thanks so much. It appears that commissioning now is successful.

wqx6, 2024-11-08:
Could you close the issue if it is resolved?

closed 2024-11-08 by dhrishi