espressif/esp-matter #1148: Matter 1.3 door lock RFID Support

bilalmalik76, 2024-11-09:
I'm working on a Matter 1.3 compatible door lock with RFID credential management. I’ve enabled RFID support through the ZAP tool and configured the feature map accordingly. However, when I test it with ecosystem apps, I don’t see any options to set or get RFID credentials.

Matter 1.3 cluster have commands to set RFID codes through client:
[image]

Issues:

None of the ecosystem apps I've tried display any options for setting or getting credentials (RFID in particular) or  ecosystems app does not support it? Any advice on getting ecosystem apps or other tools to recognize and manage RFID credentials would be greatly appreciated!

lboue, 2024-11-09:
Did you you using chip-tool?
```
chip-tool doorlock set-credential OperationType Credential CredentialData UserIndex UserStatus UserType destination-id endpoint-id-ignored-for-group-commands
```

bilalmalik76, 2024-11-10:
@lboue I am testing with different ecosystems apps how can I find if ecosystems app support it or not?

lboue, 2024-11-11:
> @lboue I am testing with different ecosystems apps how can I find if ecosystems app support it or not?

I don't believe there is a document containing this information.

dhrishi, 2025-02-10:
You will have to check the ecosystem websites/announcements for this. There isn't such a common document in the open source.

closed 2025-02-10 by dhrishi