espressif/esp-matter #1149: ICD example not starting Matter.

tomasmcguinness, 2024-11-11:
I have flashed an ESP32-H2 with the ICD example.

I get the following trace:

--- Waiting for the device to reconnect.....................
W (859) esp_matter_core: Attribute 0x00000001 on cluster 0x0000001D already exists. Not creating again.
W (869) esp_matter_core: Attribute 0x00000002 on cluster 0x0000001D already exists. Not creating again.
W (879) esp_matter_core: Attribute 0x00000003 on cluster 0x0000001D already exists. Not creating again.
W (889) esp_matter_core: Attribute 0x0000FFFC on cluster 0x0000001D already exists. Not creating again.
W (899) esp_matter_core: Attribute 0x0000FFFD on cluster 0x0000001D already exists. Not creating again.
I (921) chip[DL]: NVS set: chip-counters/reboot-count = 2 (0x2)
I (924) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
I (926) BLE_INIT: Using main XTAL as clock source
I (927) esp openthread sleep: Enable ieee802154 light sleep, the wake up source is ESP timer
I (940) OPENTHREAD: Host connection mode none
I (941) BLE_INIT: ble controller commit:[217f1bf]
W (950) BLE_INIT: BLE modem sleep is enabled
I (956) BLE_INIT: Modem BLE MAC retention initialization
W (961) BLE_INIT: Enable light sleep, the wake up source is BLE timer
I (959) phy: phy_version: 230,2, 9aae6ea, Jan 15 2024, 11:17:12
I (976) phy: libbtbb version: 944f18e, Jan 15 2024, 11:17:25
I (982) btbb_init: btbb sleep retention initialization
I (988) ieee802154: ieee802154 mac sleep retention initialization
I (992) NimBLE: GAP procedure initiated: stop advertising.

I (1001) NimBLE: Failed to restore IRKs from store; status=8

I (1006) CHIP[DL]: BLE host-controller synced
I (1026) OPENTHREAD: OpenThread attached to netif
I (1029) chip[DL]: OpenThread started: OK
I (1031) chip[DL]: Setting OpenThread device type to SLEEPY END DEVICE
I(1033) OPENTHREAD:[N] Mle-----------: Mode 0x0c -> 0x04 [rx-on:no ftd:no full-net:no]

==

I enabled all the same ICD flags under the generic_switch sample, but that appears to get stuck in the same location, with the Matter stack not starting (i.e. no advertisement for commissioning etc)

I want to make a small battery powered momentary switch.

Can you provide some advice?

shripad621git, 2024-11-13:
@tomasmcguinness , For creating a momentary switch, you can call the [generic_switch create](https://github.com/espressif/esp-matter/blob/82419a3e780f614e315b9296fbd128e61b8d265c/examples/generic_switch/main/app_main.cpp#L136) and add the [momentary feature](https://github.com/espressif/esp-matter/blob/82419a3e780f614e315b9296fbd128e61b8d265c/examples/generic_switch/main/app_main.cpp#L187) in icd app app_main.cpp in place of on_off_light and verify the switch cluster presence after testing commissioning using chip-tool
Also for the above logs, can you please specify the below details:
i) esp-matter commit id
ii) esp-idf commit id
iii) ecosystem used (chip-tool/Google Home/Apple Home)

tomasmcguinness, 2024-11-14:
Thanks, 

I have successfully created a momentary switch with two endpoints and that works well. I just want to make make it run on battery now.

For your questions:
i) 82419a3
ii) c9763f62dd
iii) chip-tool/Apple Home building and compiling on macOS.

tomasmcguinness, 2024-11-30:
I got this to work. Starting with the basic ICD sample, I moved over the IoT button code and setup a generic switch endpoint wiht a momentary switch cluster.

The only issue I ran into was the espressif/button version. The `device_hal` in `esp-matter` was tied to version 2.5, but for the PowerSave feature of the IoT button, I needed version >=3.1. Once I updated the button and enabled power save, the power consumption on my H2 dropped to 1.5mA. The button works perfectly, jumping the power consumption before it goes back to sleep!

Thanks to @shripad621git for the nudge in the right direction!

closed 2024-11-30 by tomasmcguinness