espressif/esp-matter #1151: Matter Controller - Getting data from a customized cluster command

pavel808, 2024-11-13:
I am using the `esp_matter_controller` to issue commands to a customized cluster server on a Matter end device. 

From the Matter controller side, I send the command to my end device as follows : 

```
esp_matter::lock::chip_stack_lock(portMAX_DELAY);

esp_matter::controller::cluster_command* command = new esp_matter::controller::cluster_command(nodeID, endpoint_id, custom_cluster_id, custom_command_id, commandData, chip::NullOptional, command_send_success_cb, command_send_error_cb);

esp_err_t err = command->send_command();

if (err != ESP_OK)
{
    ESP_LOGE(TAG, "ERROR! Unable to send cluster command : %d", err);
}

esp_matter::lock::chip_stack_unlock();
```
Now on my end device, when the command gets received, I have a callback like this below which gets invoked. This is all good when all I need to do is perform some write action on the device, but not send a response back to the Matter controller. 

The problem I have is when I want to read  and send data from the end device back to the Matter controller application. 

I am tying to avoid using attributes, and then having to use other Matter controller calls to read attributes (which should be kept updated on the end device even without being requested)  : 

```
static esp_err_t custom_command_callback(const ConcreteCommandPath &command_path, TLVReader &tlv_data,
                                                           void *opaque_ptr)
{
    // I would like to read data here and somehow get this back to the Matter Controller
	
}
```

wqx6, 2024-11-19:
Hi, for the command data parsing, you can follow the codes in esp_matter_rainmaker component [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L222-L244) which decodes the TLV reader to a char string.

As for the command response, You can add a simple status response by call `AddStatus()`
```
chip::app::CommandHandler *command_obj = (chip::app::CommandHandler *)opaque_ptr;
if (!command_obj) {
    ESP_LOGE(TAG, "Command Object cannot be NULL");
    return;
}
command_obj->AddStatus(command_path, chip::Protocols::InteractionModel::Status::Success);
```
But if you want to add response which has complex data field, I would suggest you to add you cluster definition xml file to `connectedhomeip/src/app/zap-templates/zcl/data-model/chip` and generate the app-common files with your xml file then it will be much easier to use `AddResponse()` with the command object.

pavel808, 2024-11-20:
@wqx6 Thank you.

Yes, for the commands which need to write data only to the end device, I am parsing the data as per the esp_matter_rainmaker component.

For the commands which need to read data i am having confusions with.

For example, to send a simple integer value back which represents a value from a sensor, are there any examples of doing this using ` command_obj->AddResponse()` ?

closed 2024-11-30 by pavel808