espressif/esp-matter #1156: Bridged Device Basic Information cluster is not displayed properly in Apple HomeKit

jzwyssig, 2024-11-17:
This issue is likely not related to esp-matter or connectedhomeip and was already discussed in the past (see [#415](https://github.com/espressif/esp-matter/issues/new?assignees=&labels=&projects=&template=issue-template.md&title=#415)). However, I would appreciate it if someone could confirm this issue.

On iOS 18.1, the **Bridged Device Basic Information Cluster** is not displayed correctly in **Apple HomeKit**. While the basic information is correctly recognized on the bridge itself, the bridged devices in Apple HomeKit always show "Unknown."

[image]

Using `chip-tool`, I am able to read the bridged device's basic information successfully, and everything is displayed correctly in Google Home as well. I also tried adding the Basic Information Cluster in addition to the Bridged Device Basic Information Cluster for each bridged device, thinking that HomeKit might be referencing the wrong cluster, but this did not resolve the issue either.

jzwyssig, 2024-11-17:
Thanks, @jonsmirl, for the fast response. However, this is exactly what I did. Each bridged device has the _Bridged Device Basic Information_ cluster, and I can confirm that I am able to read all attributes on every endpoint using `chip-tool`. Additionally, the _Bridged Device Basic Information_ cluster is displayed correctly in Google Home. Only in in Apple Home it shows "Unknown". 

The "normal" Basic Information cluster is working just fine. If I tap on the bridge in Apple Home, the data is displayed correctly.

jonsmirl, 2024-11-18:
That works for me, I see all of the fields I set in Apple Home. I am on IOS 17.7 on iPad.

jzwyssig, 2024-11-18:
Thank you for the confirmation. I am currently using iOS 18.1, and I believe this feature worked in the past. That’s why I’m curious if the issue might be related to iOS 18.1. I’ve already reset the device and re-paired it multiple times, but unfortunately, it still doesn’t work.