espressif/esp-matter #1157: build step for 'chip_gn' failed No such file or directory

North3rnL1ght, 2024-11-17:
**Describe the bug**
I downloaded and installed ESP-IDF and ESP-Matter for [apple/swift-matter-examples](https://github.com/apple/swift-matter-examples). I used the installation tutorial they provided. But for some reason I can't build the example for the smart-light. I even tried the examples provided inside the esp-matter folder but I always recive the following error during Build. 

```
[702/1257] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /Users/account/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /Users/account/esp-matter/examples/light/build/esp-idf/chip/lib/libCHIP.a 
cd "/Users/account/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/account/esp-matter/examples/light/build/esp-idf/chip" && ninja esp32
/bin/sh: line 0: cd: /Users/account/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/account/esp-matter/examples/light/build/esp-idf/chip: No such file or directory 
```

**Environment**
- ESP-Matter Commit Id: 7b9d69aa0a21735b6016808a000210bdf50c9ccb (grafted, HEAD -> release/v1.2, origin/release/v1.2)
- ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb (grafted, HEAD, tag: v5.2.1)
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6

- Host Machine OS: MacOS 15.1 (24B83)
- Host Machine Python version: 3.9.6

**Any additional details**
The Problem is that the build.ninja trys to execute the following command:

```
#############################################
# Custom command for esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build

build esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a | ${cmake_ninja_workdir}esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build ${cmake_ninja_workdir}esp-idf/chip/lib/libCHIP.a: CUSTOM_COMMAND esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure || esp-idf/chip/args_gn
  COMMAND = cd "/Users/account/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/account/esp-matter/examples/light/build/esp-idf/chip" && ninja esp32
  DESC = Performing build step for 'chip_gn'
  restat = 1
```

Which results in "No such file or directory". When I manually edit the build.ninja file and remove the first PATH "/Users/account/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;" the build executes successfully. I don't know where it takes the first path from. Even tried reinstalling esp-matter and esp-idf but it didn't made any changes.

shubhamdp, 2024-11-19:
@North3rnL1ght thanks for the detailed debugging, will see how this can be fixed.

On my mac, I am unable to reproduce this. I only see one path in `build.ninja`, not sure how that extra path added.
I'm using following gn, ninja, cmake, versions, can you please share your's.
```
$  esp-matter git:(7b9d69a) gn --version
2123 (5d8727f3fbf4)
$  esp-matter git:(7b9d69a) cmake --version
cmake version 3.30.0
$  esp-matter git:(7b9d69a) ninja --version
1.12.1
```

North3rnL1ght, 2024-11-19:
@shubhamdp I found this extra path inside the list of component path output on set-target for esp32c6.

Copied only some until the components/chip folder path occours!
```
-- Component paths: /Users/account/esp-matter/examples/common/app_bridge /Users/account/esp-matter/examples/common/app_reset /Users/account/esp-idf/components/app_trace /Users/account/esp-idf/components/app_update /Users/account/esp-idf/components/bootloader /Users/account/esp-idf/components/bootloader_support /Users/account/esp-idf/components/bt /Users/account/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip
```

The only different version is in cmake:
```
gn --version
2123 (5d8727f3fbf4)
cmake --version
cmake version 3.31.0
CMake suite maintained and supported by Kitware (kitware.com/cmake).
ninja --version
1.12.1
```

nicolas-christe, 2024-11-21:
I have the same issue, fixed by rolling back cmake from 3.31.0 to 3.30.5.

shubhamdp, 2024-11-22:
@nicolas-christe thanks for the tip. Please check if https://github.com/project-chip/connectedhomeip/pull/36606 can fix the problem with cmake v3.31.x. You can apply the PR by running following command in esp-matter.

```
curl https://patch-diff.githubusercontent.com/raw/project-chip/connectedhomeip/pull/36606.patch | git -C connectedhomeip/connectedhomeip am
```

nicolas-christe, 2024-11-22:
Thanks @shubhamdp, I confirm that the patch works for me with both cmake 3.30.5 and 3.31.0.

North3rnL1ght, 2024-11-22:
@shubhamdp thank you very much. Patch works for me as well on cmake 3.31.0 👍

closed 2024-12-04 by espressif-bot

shubhamdp, 2024-12-04:
We have rolled up the connectedhomeip submodule in 0e53bcab756254026df2406a39f391ef5a7ffd2c which has the patched fix.