espressif/esp-matter #1158: esp_matter_controller pairing queries

pavel808, 2024-11-18:
I am using the new `pairing_code_thread() `API which works well. I have two questions : 

1) How do I assign a callback to get the commissioning status (success or fail) when calling that particular `pairing_code_thread() `function? I see that there is a `pairing_command` class which can be used to assign callbacks when an object is created, but doesn't yet have parameters for working with pairing codes or QR codes. 

My guess is that this is still under development? 

2) Is there a n API implemented for unpairing / de-commissioning a device yet? 

Thank you in advance.

wqx6, 2024-11-19:
Hi, there are some callbacks in the pairing_command class to notify whether the pairing and commissioning are successful/failed. When the PASE session is established or when there is an error during PASE session establishment the `OnPairingComplete` will be called. When the commissioning is completed, `OnCommissioningComplete` will be called. but currently there are still some callbacks not implemented and we still need some APIs for the users to set their custom callbacks. We will add them later

For the second question, you can use `controller_instance.get_commissioner()->UnpairDevice(peerNodeId)` to unpair a device. The commissioner will remove the fabric of the remote device in that function.

pavel808, 2024-11-20:
> Hi, there are some callbacks in the pairing_command class to notify whether the pairing and commissioning are successful/failed. When the PASE session is established or when there is an error during [...]

@wqx6 Thanks again. Currently I am not creating a `pairing_command` object and  just calling the `pairing_code_thread()` function which is not part of the `pairing_command` class. 

I guess the `pairing_command` class has yet to be updated to have a pairing code / QR code field ? 

Alternatively, can I still create a `pairing_command` object anyways, but call the `pairing_code_thread()` function separately and still have the callbacks updated as expected.

wqx6, 2024-12-18:
@pavel808 Hi, we have supported setting callbacks for pairing commands. You can call `pairing_command::get_instance().set_callbacks()` before using `pairing_code_thread()`. Note that the pairing_command is a singleton, so you cannot create another object for it. The `pairing_command` class only acts as the delegate for device discovery and device pairing, so it will not maintain fields for code / QR code.

pavel808, 2024-12-18:
@wqx6 Perfect. Great timing. Thanks. 

What about for decommissioning / unpairing ? Currently I just call `get_commissioner()->UnpairDevice() `without any notifications.

pavel808, 2025-01-02:
@wqx6 I guess i should close this ticket. The callbacks get invoked as expected with the correct parameters after commissioning. Thanks. 

Can I just ask if there is something similar for de-commissioning / un-paring yet ? User callbacks that can be assigned. 

For un-pairing I call `get_commissioner()->UnpairDevice()` , but no callbacks get assigned or invoked to let me know if the operation was successful or not

closed 2025-02-10 by dhrishi