espressif/esp-matter #1159: How to add and remove supporterd attributes of a cluster though esp-matter without using ZAP

bilalmalik76, 2024-11-18:
How to add and remove supporterd attributes of a cluster though esp-matter without using ZAP? Considering esp-matter provide non-zap based APIs, how to find enabled attributes information and how to enable and disbale specific?

shubhamdp, 2024-11-19:
To check if attribute/cluster present or not there are `get()` APIs. If the API returns `nullptr` then the attribute is not present on the node.

For cluster: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L397-L407
For Attribute: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L567-L578

To add, we have `::create()` APIs listed in [esp_matter_attribute.h](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.h).

We do not have APIs to remove the already present attributes. But, we only add the mandatory attributes/events when creating the device type, so I think you should consider only adding the optional clusters/attributes/events.

bilalmalik76, 2024-11-19:
@shubhamdp What I understood is that all mandatory attributes for a specific cluster of a sepcific device type are already available in esp-matter and we only need to add optional if required?

shubhamdp, 2024-11-19:
@bilalmalik76 Yes, thats right. When adding new device type, we ensure and only add the mandatory bits from the specification.

bilalmalik76, 2024-11-19:
Thanks where I can find the list of mandatory attributes available in esp-matter?

shubhamdp, 2024-11-19:
[esp_matter_cluster.cpp](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp) contains the list of attributes/commands/events which are mandatory for a cluster.

shubhamdp, 2024-11-28:
@bilalmalik76 any updates on this? Please close the issue if your question is answered.

bilalmalik76, 2024-11-28:
I am closing this and will reopen if needed.

closed 2024-11-28 by bilalmalik76