espressif/esp-matter #1160: Build fails with ERROR: Failed building wheel for cffi

evanthomas, 2024-11-19:
**Describe the bug**
I'm trying to set up Matter according to the docs here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#esp-matter-setup

./install.sh fails with 
```
ERROR: Failed building wheel for cffi
```

**Environment**
- ESP-Matter Commit Id: a3dbd43d8c9bca3ba234d5de1de497ca9abcaadb
- ESP-IDF Commit Id: 87c9fb8d405c05e7cf74976040b8887f8265c0b8
- SoC (eg: ESP32 or ESP32-C3): ESP32-H2
- Device Logs (Please attach the log file): N/A

- Host Machine OS: Mac OS 15.1
- Host Machine Python version: 3.13.0

- Commissioner app and versions if present:
- Home hub app and versions if present: N/A
- Commissioner's logs if present: N/A

**Any additional details**
```
Running Matter Setup

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (16.0s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....done (1m5.8s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20241119 14:38:28 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...
[... 96 lines cut ...]
      clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/ffi -I/Users/evan/.espressif/python_env/idf5.3_py3.13_env/include -I/usr/local/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13 -c c/_cffi_backend.c -o build/temp.macosx-14.0-x86_64-cpython-313/c/_cffi_backend.o -iwithsysroot/usr/include/ffi
      c/_cffi_backend.c:4517:22: warning: 'Py_FileSystemDefaultEncoding' is deprecated [-Wdeprecated-declarations]
       4517 |                      Py_FileSystemDefaultEncoding, &filename_or_null, &flags))
            |                      ^
      /usr/local/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13/fileobject.h:22:1: note: 'Py_FileSystemDefaultEncoding' has been explicitly marked deprecated here
         22 | Py_DEPRECATED(3.12) PyAPI_DATA(const char *) Py_FileSystemDefaultEncoding;
            | ^
      /usr/local/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13/pyport.h:251:54: note: expanded from macro 'Py_DEPRECATED'
        251 | #define Py_DEPRECATED(VERSION_UNUSED) __attribute__((__deprecated__))
            |                                                      ^
      c/_cffi_backend.c:6100:9: error: call to undeclared function '_PyErr_WriteUnraisableMsg'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
       6100 |         _PyErr_WriteUnraisableMsg(PyText_AS_UTF8(s), NULL);
            |         ^
      c/_cffi_backend.c:6100:9: note: did you mean 'PyErr_WriteUnraisable'?
      /usr/local/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13/pyerrors.h:233:18: note: 'PyErr_WriteUnraisable' declared here
        233 | PyAPI_FUNC(void) PyErr_WriteUnraisable(PyObject *);
            |                  ^
      1 warning and 1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cffi
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (cffi)
...

shubhamdp, 2024-11-19:
[cffi v1.15.1](https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/constraints.txt#L35-L36) fails when using python3.13 (https://github.com/python-cffi/cffi/issues/23) and Python 3.13 support is added onwards v1.17.0.

Can you please try if this works for you, and let me know, I'll add this change to the connectedhomeip
```
# on macos
sed -i '' 's/cffi==1.15.1/cffi==1.17.1/g' connectedhomeip/connectedhomeip/scripts/setup/constraints.txt

# on linux
sed -i 's/cffi==1.15.1/cffi==1.17.1/g' connectedhomeip/connectedhomeip/scripts/setup/constraints.txt
```

evanthomas, 2024-11-19:
Unfortunately, is still fails.

```
➜  esp-matter git:(main) sed -i '' 's/cffi==1.15.1/cffi==1.17.1/g' connectedhomeip/connectedhomeip/scripts/setup/constraints.txt
➜  esp-matter git:(main) cd ./connectedhomeip/connectedhomeip
➜  connectedhomeip git:(ce0101d2) ✗ ./scripts/checkout_submodules.py --platform esp32 darwin --shallow
Checking out: nlassert, nlio, mbedtls, qrcode, m5stack-tft, pigweed, openthread, nanopb, freertos, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, third_party/libwebsockets/repo, perfetto, third_party/abseil-cpp/src, third_party/fuzztest, third_party/googletest, third_party/re2/src
➜  connectedhomeip git:(ce0101d2) ✗ cd ../..
➜  esp-matter git:(main) ./install.sh 
Running Matter Setup

2024-11-19 17:30:12,526 Loading extra packages for darwin
2024-11-19 17:30:12,527 Appending: /Users/evan/projects/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json for this platform
2024-11-19 17:30:12,527 Skipping: windows (i.e. /Users/evan/projects/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (42.6s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....done (1m33.6s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20241119 17:32:40 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...

[notice] A new release of pip is available: 23.2.1 -> 24.3.1
[notice] To update, run: pip install --upgrade pip
Building host tools

Done. Made 9150 targets from 412 files in 500ms
ninja: Entering directory `/Users/evan/projects/esp/esp-matter/connectedhomeip/connectedhomeip/out/host'
[128/941] ar libSystemLayer.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: obj/src/system/lib/libSystemLayer.a(libSystemLayer.SystemMutex.cpp.o) has no symbols
[416/941] ar boringssl.a
[... 387 lines cut ...]
    # via
    #   -r requirements.all.txt
    #   pip-tools

# The following packages are considered to be unsafe in a requirements file:
pip==23.2.1
    # via
    #   fastcore
    #   ghapi
    #   pip-tools
setuptools==68.0.0
    # via
    #   pip-tools
    #   west

# Manual edits:

# Higher versions depend on proto-plus, which break
# nanopb code generation (due to name conflict of the 'proto' module)
google-api-core==2.17.0
➜  esp-matter git:(main) grep cffi ./connectedhomeip/connectedhomeip/scripts/setup/constraints.txt
cffi==1.17.1
    # via cffi
➜  esp-matter git:(main) 
```

shubhamdp, 2024-11-19:
Can you try this patch: https://github.com/project-chip/connectedhomeip/pull/36549/files. This won't be applied cleanly, you will have to manually edit the files.

shubhamdp, 2024-11-19:
Please try removing the earlier environment: `rm -rf connectedhomeip/connectedhomeip/.environment`

shubhamdp, 2024-11-20:
@evanthomas matter environment is setup correctly, this is the last step, and this cffi is from esp-matter-mfg-tool.
If you are not planning to use esp-matter-mfg-tool then you can continue to build examples.

We have identified and will push fix till then can you please remove that dependency from https://github.com/espressif/esp-matter/blob/b5ed1c30e2fb8d38066057ab1f16438728d82939/requirements.txt#L11

evanthomas, 2024-11-20:
Works for me - thanks.

shubhamdp, 2024-11-22:
@evanthomas If you could please pull the latest esp-matter and re-run the install.sh, we have updated the submodule  in e09e20b693d1921f997f373055f92fb04e395fb1 to work with Python3.13.

evanthomas, 2024-11-22:
I can confirm it builds without error now.

closed 2024-11-28 by espressif-bot

shubhamdp, 2024-11-28:
@evanthomas thanks for the confirmation, I've also bumped the esp-matter-mfg-tool version which supports Python v3.13, and that commit (a6c9e5b9c72e3a89a700d7650344f92389c71f58) closed this issue.