espressif/esp-matter #1161: Reading the Thread TLVS dataset

pavel808, 2024-11-19:
It's possible to read the Thread TLVS dataset with the following Matter esp cli command : 

`matter esp ot_cli dataset active -x `

How is it possible to read this value from within application code? \

I have a C++ Matter controller application running on the ESP Thread BR device and currently I am hard-coding the dataset tlvs value there for commissioning. It would be much better to be able to read this directly from the TBR into the application instead if possible. Thanks in advance.

shubhamdp, 2024-11-19:
You can get the active dataset using `otDatasetGetActiveTlvs()` API.

Please find the documentation here: https://openthread.io/reference/group/api-operational-dataset#otdatasetgetactivetlvs

Please find the usage here (this is what gets executed when you hit `matter esp ot_cli dataset active -x`): https://github.com/openthread/openthread/blob/main/src/cli/cli_dataset.cpp#L698

shubhamdp, 2024-11-28:
@pavel808 any updates on this? Please close the issue if your question is answered.

pavel808, 2024-11-28:
> @pavel808 any updates on this? Please close the issue if your question is answered.

Apologies for the delay. This solves my problem. Many thanks.

closed 2024-11-28 by pavel808