espressif/esp-matter #1164: BLE bonding fail when it is no longer used by Matter

GaryChan-work, 2024-11-20:
Hi,

I had the same issue as listed below and could use @chrelu solution to do the advertising. However, I couldn't perform the bonding with the mobile phone as I did before. I have used the "nRF Connect app" on Android to connect it and perform the bonding, but the connection was dropped, and the bonding was failed. The mobile didn't prompt any message for me to confirm the passkey, and the device also didn't receive BLE_GAP_EVENT_PASSKEY_ACTION as before.

Does anyone know how to fix this issue?
Thank you very much!

              I found a solution. Instead of re-initializing BLE, I only reset BLE GATTS and start it again after adding my BLE services (see method gatt_svr_init). I hope I could also help someone else.

```c
void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type)
    {
    ...
    case chip::DeviceLayer::DeviceEventType::kCHIPoBLEConnectionClosed:
        ble_start();
        break;
    ...
    }
}
``` 
```c
void ble_start(void)
{
    ble_hs_cfg.sync_cb = sync_cb;
    ble_hs_cfg.reset_cb = reset_cb;

    gatt_svr_init();

    sync_cb();
}
``` 
```c
void gatt_svr_init()
{
  ble_gatts_reset(); // before I didn't call this method

  ble_svc_gap_init();
  ble_svc_gatt_init();
  ble_gatts_count_cfg(gatt_svr_svcs);
  ble_gatts_add_svcs(gatt_svr_svcs);

  ble_gatts_start(); // before I didn't call this method
}
``` 
```c
void sync_cb(void) {
  ble_hs_id_infer_auto(0, &addr_type);
  advertise();
}
```

_Originally posted by @chrelu in https://github.com/espressif/esp-matter/issues/906#issuecomment-2077325524_

shubhamdp, 2024-12-18:
@GaryChan-work extremely sorry for the late turn around. Can you please check if https://github.com/espressif/esp-matter/issues/1207#issuecomment-2547741523 answers your problem.

For continue using the BLE even after commissioning, we need to disable the `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` option.

What I tried doing is used the code from the esp-idf examples' bleprph and blecent to incorporate those roles.

Please check and let me know if that works for you.

shubhamdp, 2024-12-18:
Please check if https://github.com/espressif/esp-idf/issues/11036 is helpful.

dhrishi, 2025-01-10:
@GaryChan-work Please close this if your issue is resolved

dhrishi, 2025-03-18:
Closing this now

closed 2025-03-18 by dhrishi