espressif/esp-matter #1166: How do I determine the clusters required for a test product certification to determine the PICS file

2522957270, 2024-11-20:
How do I determine the clusters required for a test product certification to determine the PICS file?The above is preparation for a new product. My ultimate goal is to test CASE (matter authentication) through determined PICS files, so I want to know the official method to confirm endpoint, cluster, PICS, and CASE. Ask for help. Thank you！

wqx6, 2024-11-21:
You should determine the clusters for your device according to the SPEC. The default APIs in esp-matter will add clusters that are mandatory for the specific device. After creating the device endpoint, you can add optional clusters/attributes/commands to the device endpoint if your product needs them. After that you can run the [PICS-generator](https://github.com/project-chip/connectedhomeip/tree/master/src/tools/PICS-generator) to generate the PICS for the device.

wqx6, 2024-11-21:
Note that `TC-IDM-10.2` will check the cluster conformance for features, attributes and commands across all endpoints. You can run that test case to check whether there are some clusters/features/attributes/commands missed for your device.

2522957270, 2024-11-21:
Thank you, but there's more.My development environment is to pull down docker images in ubuntu (docker pull espressif/esp-matter:release-v1.3).But I can't seem to use the PICS-generator tool (perhaps due to a lack of dependencies or outdated version), how can I fix this?
And I have a few more questions:
1. How to generate PICS files: Run scripts/build_python.sh to generate all PICS files, and then export PICS files required for authentication from all PICS files through PICS-generator?
2. Is the PICS file required for authentication exported by code or a burned device?
3. Is there an example of the above process?

closed 2024-11-22 by 2522957270