espressif/esp-matter #1169: Give a option to make on_level

Haerteleric, 2024-11-25:
**Is your feature request related to a problem? Please describe.**
On Level Attribute  (Level Control Cluster) is reset to init values after each power cycle.

**Describe the solution you'd like**
allow ATTRIBUTE_FLAG_NONVOLATILE for that attribute.

**Describe alternatives you've considered**
give us a esp_matter func to manually set the ATTRIBUTE_FLAG_NONVOLATILE of a Attribute

**Additional context**
...

jonsmirl, 2024-11-25:
Current Level is required by the spec to be non-volatile.
[image]

I just tried the Espessif Light Demo and it does not appear to be preserving CurrentLevel like it is supposed to do.
You don't need option, this appears to be broken in esp-matter.

Haerteleric, 2024-11-27:
i am not talking about the "CurrentLevel" (0x0000) but the "OnLevel" (0x0011) created in 

`esp_matter::cluster::level_control::attribute::create_on_level`
(esp_matter_attribute.cpp:1136)

Haerteleric, 2024-11-27:
why should OnLevel be Volatile ?

That does not make sense. If a user sets a TurnOn Brigthness (not startup brightness) it should default to Max Brightness everytime the device gets power cycled.

Haerteleric, 2024-11-27:
see 1.6.4.1.1. Effect of On/Off Commands on the CurrentLevel Attribute

matter spec

shubhamdp, 2024-11-28:
@Haerteleric I think you should be using `StartUpCurrentLevel` and not `OnLevel`

> 1.6.6.15. StartUpCurrentLevel Attribute
This attribute SHALL indicate the desired startup level for a device when it is supplied with power and this level SHALL be reflected in the CurrentLevel attribute.

shubhamdp, 2024-12-10:
@Haerteleric any update on the issue?

Haerteleric, 2024-12-12:
The `StartUpCurrentLevel` is for power-on-behaviour  imo.

The `OnLevel` is when the light EP was `Off` (but device is powered and on Network) and got turned on by a `toggle` or `turn on` command.

shubhamdp, 2024-12-12:
@Haerteleric That is correct, and you should use `StartUpCurrentLevel` for re-storing level value on power cycle. I think that was the original question you asked in the description.

Is there still a problem? If not can you please close the issue.

Haerteleric, 2024-12-12:
`OnLevel` is volitile though and has to be set after every PowerCycle atm. I don't this this is the way it should behave.

Haerteleric, 2024-12-12:
[image]
[image]

Haerteleric, 2024-12-12:
the stack resets the `OnLevel` value after every reboot to the default setting.

shubhamdp, 2024-12-12:
> the stack resets the `OnLevel` value after every reboot to the default setting.

That is expected, as per the specification, `OnLevel` is a volatile attribute. If you want to mess with spec compliance please add `ATTRIBUTE_FLAG_NONVOLATILE` below.

https://github.com/espressif/esp-matter/blob/cb216647d443444af8cc858a009228849e70287f/components/esp_matter/esp_matter_attribute.cpp#L189-L195

NOTE: I won't recommend you to make that change, but if you want to break the specification compliance, please go ahead.

Haerteleric, 2024-12-12:
Could you please refer me to the specification sections where this is defined? 

Cause i can't find it.

Haerteleric, 2024-12-12:
the code section you linked is about the node_label ?!?

> That is expected, as per the specification, `OnLevel` is a volatile attribute. If you want to mess with spec compliance please add `ATTRIBUTE_FLAG_NONVOLATILE` below. [...]

shubhamdp, 2024-12-12:
Sorry for the cut-paste mistake, I have updated the link in the comment above.

Please check Matter Application Cluster Specification Section 1.6.6. Attributes. Do check the Quality Column, `X` is nullable and `N` is non-volatile.

Haerteleric, 2024-12-12:
You are right. I was in the wrong. 

I would like to apologize for my tone.

closed 2024-12-12 by Haerteleric