espressif/esp-matter #1170: GroupCluster allways reports GroupNameSupport even if config is set to false

Haerteleric, 2024-11-25:
**Describe the bug**

```c
cluster::groups::config_t groupsConfig;
groupsConfig.group_name_support = false;
cluster_t *groupCluster = cluster::groups::create(endpoint, &groupsConfig, CLUSTER_FLAG_SERVER);
```

should result in 

```
global::attribute::create_feature_map(cluster, 0);
```

but in esp_matter::cluster::groups::create() (`esp_matter_cluster.cpp:1233`)

it is hardcoded to:

```
global::attribute::create_feature_map(cluster, 1);
```

Haerteleric, 2024-11-25:
This causes issues at Matter Cert Lab. As the PICS files are incorrect because of it.

jonsmirl, 2024-11-25:
Why do you want to turn it off? It only consumes around 50 bytes of flash, and I like giving my groups human friendly names.

jadhavrohit924, 2024-11-26:
@Haerteleric If you check the latest spec, `GroupNameSupport` is mandatory on the groups cluster. BTW, which branch/commit you are using?

Haerteleric, 2024-11-27:
i agreen, that there is no obvious reason to disable that. But if disabled the featureMap should be set accordingly.

jadhavrohit924, 2024-11-27:
Yes, that’s why I asked you for the branch/commit that you are using.
> groupsConfig.group_name_support = false;

I'm not sure where you got that. If you check the latest main branch we don’t have any such config in the groups cluster. If it's an attribute that you are referring you cannot set it to false. To get a better understanding of your issue, I would need a branch/commit id that you are using. Will it be possible to share?

Haerteleric, 2024-11-27:
im using the esp-matter as a component (v1.3.0) on esp-idf v5.4.0 (46acfdce969f03c02b001fe4d24fa9e98f6adc5e).

the groupsConfig is a extended_color_light::config_t struct.
`esp_matter_endpoint.h:219`

i just checked the same file in the main branch and it has indeed been reworked.

jadhavrohit924, 2024-11-27:
I was talking about the `group_name_support` under the groups cluster. It's not in the SDK, what is `group_name_support` that you have set to `false`. Please share the details of the `group_name_support`. Did you add any patch on top of esp-matter? If yes, please share it with me.

Haerteleric, 2024-11-28:
I'm using https://components.espressif.com/components/espressif/esp_matter

this uses the 1.3 Branch of this repo. via https://github.com/espressif/esp-matter/. 

in my project i create a extended color light EP using this struct:
https://github.com/espressif/esp-matter/blob/552b6f8e9294a5d35dcd943f09c54edcca3269ab/components/esp_matter/esp_matter_endpoint.h#L223

the flag in question can be found here :

https://github.com/espressif/esp-matter/blob/552b6f8e9294a5d35dcd943f09c54edcca3269ab/components/esp_matter/esp_matter_cluster.h#L305

jadhavrohit924, 2024-11-28:
Okay, so `group_name_support` here doesn’t represent a feature, and it's of type `bitmap8` in the spec, so you cannot set it to false. In general, features represent attributes, commands, and events, not vice versa.

shubhamdp, 2024-11-28:
I'd suggest you to toggle that feature bit in the PICS file.

shubhamdp, 2024-12-10:
@Haerteleric any update on this, if your question is answered can you please close the issue

Haerteleric, 2024-12-12:
yes we ended up enabling it, and setting it in the pics file. (We passed certification)

this was not really an issue just wanted to inform you about the dummy nature of that config flag

closed 2024-12-12 by Haerteleric