espressif/esp-matter #1172: The use of PICS-generator

2522957270, 2024-11-27:
I want to use PICS-generator to generate the PICS files required for device authentication, and this is an error that occurs during use.How to solve?
[image]

VaishaliAvhale, 2024-11-27:
Please provide the complete logs from the DUT and the PICSGenerator script. Additionally, could you specify the version for which you are trying to generate the PICS?

2522957270, 2024-11-28:
ESP32C6;This is my printed log, similar to the one in the picture;Yes, I want to do matter-v1.3 certification;
[PICS-generator.log](https://github.com/user-attachments/files/17942297/PICS-generator.log)

VaishaliAvhale, 2024-11-29:
@2522957270,

The PICSGenerator tool development is still in progress, and hence it is not currently recommended for use as per CSA's guidance. Please generate the PICS manually for now. Once the tool is ready, we will notify you.

2522957270, 2024-11-29:
> @2522957270,
> 
> The PICSGenerator tool development is still in progress, and hence it is not currently recommended for use as per CSA's guidance. Please generate the PICS manually for now. Once the tool is ready, we will notify you.

How to generate PICS manually? Do you manually select the PICS required for the development device by referring to the PICS required for other certified devices?

2522957270, 2024-11-29:
How to generate PICS manually? Do you manually select the PICS required for the development device by referring to the PICS required for other certified devices?

VaishaliAvhale, 2024-12-02:
Please refer this document to generate PICS file manually: 
[PICS_Generation_Guide.pdf](https://github.com/user-attachments/files/17972056/PICS_Generation_Guide.pdf)

closed 2024-12-02 by 2522957270