espressif/esp-matter #1174: Restart BLE advertising after device is de-commissioned

pavel808, 2024-11-27:
I am working on a Matter end device application on an ESP32-C6. 

Whenever a Matter controller unpairs / de-commissions the end device from the Matter / Thread network, it doesn't start to advertise again over BLE again until I re-flash the device. 

On the Matter Controller side I am calling `controller_instance.get_commissioner()->UnpairDevice(devices.at(device_idx).nodeID)`. It appears from the logs that the fabric gets removed. 

However, what is the best way to force the end device to re-start advertising over BLE again after the fabric is removed, so that it can be commissioned again? Thanks.

DejinChen, 2024-11-28:
Hi, to start BLE advertising, you can invoke `esp_matter::factory_reset()` directly when received `DeviceEventType::kFabricRemoved` event and the count of fabric is zero. 
Or you can change the parameter `kDnssdOnly` to `kAllSupported` for function `OpenBasicCommissioningWindow`.  But this option only works when config `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` is disabled.

pavel808, 2024-12-03:
> Hi, to start BLE advertising, you can invoke `esp_matter::factory_reset()` directly when received `DeviceEventType::kFabricRemoved` event and the count of fabric is zero. Or you can change the [...]

@DejinChen Thanks or this. Changing  `kDnssdOnly `to `kAllSupported `works in my case perfectly, and I can then commission again. 

Regarding the timeout parameter for `OpenBasicCommissioningWindow`, if I set this to zero, then the commissioning window should not close at all?

shubhamdp, 2024-12-03:
@pavel808 according to Core Matter specification, Section `5.4.2.3. Announcement Duration`, there is minimum and maximum duration for advertisement. Minimum is 3 min and max is 15 and in case of extended advertisement, it can be extended to 48 Hrs.

pavel808, 2024-12-03:
> @pavel808 according to Core Matter specification, Section `5.4.2.3. Announcement Duration`, there is minimum and maximum duration for advertisement. Minimum is 3 min and max is 15 and in case of extended advertisement, it can be extended to 48 Hrs.

@shubhamdp Ok thanks. 

It's a bit off topic, but let's take the Matter light example, for example.

The Matter core gets started initially with the call :` esp_matter::start(app_event_cb)` . 300 seconds is the default advertising window duration set within the SDK. 

How can we set the advertising window duration from our application initially before any `kFabricRemoved` event ?

shubhamdp, 2024-12-04:
@pavel808 SDK uses the 15 minutes timeout and not 300 seconds, and if the extended advertisement is enabled then it is 48 Hrs.

If you'd like to not open the commissioning window on boot, you can disable [CONFIG_ CHIP_ENABLE_PAIRING_AUTOSTART](https://github.com/shubhamdp/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L319-L324) option and open the commissioning window as per your convenience.

shubhamdp, 2024-12-10:
@pavel808 can you please close the issue if your question is answered.

closed 2024-12-10 by pavel808