espressif/esp-matter #1175: The curtains under the bridge device keep showing "Close..." in Apple Home Control. The specific percentage cannot be displayed. What is the reason?

wolen666, 2024-11-28:
I used the bridge-app under the official matter1-2 to dynamically create a curtain node test code. After issuing the specific percentage command, I waited for 6 seconds and then uploaded the value of the OperationalState attribute. The ideal effect should be to display "Close..." first and then display the current specific percentage. In fact, in Apple Home, it always shows that it is shutting down... I don't know where is the problem in the code. Below is my main code attachment.

https://github.com/user-attachments/assets/754f854d-701b-408d-9629-4814211ff418

[win.txt](https://github.com/user-attachments/files/17946386/win.txt)

shubhamdp, 2024-11-28:
can you please share the patch that I can apply on top of `connectedhomeip/examples/bridge-app/esp32` so that we can give a try.

Please share the esp-idf used during this demonstration. Assuming esp-matter used here is 7b9d69aa0a21735b6016808a000210bdf50c9ccb i.e. latest from release/v1.2 branch. (please share the correct esp-matter revision if this is incorrect)

wolen666, 2024-11-29:
[win-app.zip](https://github.com/user-attachments/files/17956098/win-app.zip)
win-app.zap:Is that the bridge contains a curtain.
Apple Family Pairing Code:20202021
Matter1-2 branch info:
      commit 13028cf6bf21f211215ed513e5e03d78b46418eb (HEAD -> v1.2-branch, origin/v1.2-branch)
      Author: shripad621git <79364691+shripad621git@users.noreply.github.com>
      Date:   Mon May 27 18:27:14 2024 +0530
      Adding config checks for functions specific to only wifi and thread in NetworkCommissioning Cluster code. (#33268) 
      (#33606) 
      * Adding config checks for fucntions specific to only wifi and thread in NetworkCommissioning
      * Restlyed
esp-idf branch info:
    commit e3e112e6122dfd064a78f3dece8cdce17a472e13 (HEAD -> release/v5.1, origin/release/v5.1)
    Merge: c1f26cca62 910c82b27c
    Author: Martin Vychodil <martin@espressif.com>
    Date:   Mon Mar 4 20:48:45 2024 +0800
    Merge branch 'fix/examples_format_sd_card_disable_v5.1' into 'release/v5.1'
    fix(examples): Disable formatting SD cards in storage examples by default (v5.1)
    See merge request espressif/esp-idf!29004

Thank you, I look forward to your follow-up reply！

pimpalemahesh, 2024-12-03:
@wolen666 We tested your changes and observed that the UI on Apple Home is not updating after setting the window cover to a specific percentage. However, we tested the same changes with the Google ecosystem, and it is working as expected. Since this issue appears to be on Apple Home's side, we will report it to them soon.

wolen666, 2024-12-03:
I think it may not be a problem with Apple Home, because I saw that there are matter curtain products on the market. His control logic is: 

First, the opening percentage of the Apple UI is set, and then the device opens the curtains after receiving the instruction. The Applehome UI will display that it is opening..., 

And then the curtains move to the set percentage before uploading the information. Go to AppleHome, and finally the UI changes from opening... to a specific percentage. 

What I'm more confused about now is what are the uploaded attribute values ​​of "upload information to AppleHome after the curtains move to a set percentage." 

I tried a lot of attributes but still couldn't change the UI from "opening... to a specific percentage". Below is a link to find the finished curtains from youtube. They are normal.（4:06 of video）

https://www.youtube.com/watch?v=S1HAjicQ8R8

pimpalemahesh, 2024-12-04:
@wolen666 
Could you please update your HandleDeviceStatusChanged() callback as follows:

```
void HandleDeviceStatusChanged(Device * dev, Device::Changed_t itemChangedMask)
{
    if (itemChangedMask & Device::kChanged_Reachable)
    {
        ScheduleReportingCallback(dev, BridgedDeviceBasicInformation::Id, BridgedDeviceBasicInformation::Attributes::Reachable::Id);
    }

    if (itemChangedMask & Device::kChanged_Windows)
    {
        ScheduleReportingCallback(dev, WindowCovering::Id, WindowCovering::Attributes::CurrentPositionLiftPercent100ths::Id);
    }

    if (itemChangedMask & Device::kChanged_Name)
    {
        ScheduleReportingCallback(dev, BridgedDeviceBasicInformation::Id, BridgedDeviceBasicInformation::Attributes::NodeLabel::Id);
    }
}
```
Let me know if this resolves the issue.

wolen666, 2024-12-04:
I just tried it and I am very happy. This modification can indeed solve the problem. Thank you very much.

dhrishi, 2024-12-04:
Can you please close this and create a new issue for the above problem? Ideally, I would suggest to report this to Apple through the feedback assistant?

closed 2024-12-04 by wolen666