espressif/esp-matter #1177: TC-IDM-10.2,ScenesManagement: The certification body feedback that this is a temporary cluster.

ethan-jy, 2024-11-29:
**Describe the bug**
ScenesManagement: The certification body feedback that this is a temporary cluster.
The certification body told me to close it in the program, what should I do

**Environment**
- ESP-Matter Commit Id:origin/release/v1.3
- ESP-IDF Commit Id:5.3.1
- SoC (eg: ESP32 or ESP32-C3):ESP32H2
- Device Logs (Please attach the log file):

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

VaishaliAvhale, 2024-12-02:
Please add the following line to the test_parameter section of the Test Harness project configuration:
`"bool-arg": "all_provisional:true"`
Once updated, kindly re-run the test case.

dhrishi, 2024-12-23:
@ethan-jy Please close the issue if resolved

dhrishi, 2025-03-03:
Closing this now. Please re-open in case of any issues

closed 2025-03-03 by dhrishi