espressif/esp-matter #1178: TC-IDM-10.1;contains duplicate values  spec_location: Global Elements；Contains two 65530.

ethan-jy, 2024-11-29:
**Describe the bug**
[MatterTest] 11-27 02:48:35.950 INFO
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_1
    location:
        Endpoint: 0,
        Cluster:  40 (0x28) BasicInformation,
        Attribute:65531 (0xfffb)
    problem: Failed validation of value on Endpoint 0, Cluster BasicInformation (40, 0x0028), Attribute AttributeList 65531, 0xFFFB: Value [65528, 65529, 65530, 65531, 65532, 65530, 0, 6, 1, 2, 3, 4, 7, 8, 9, 10, 19, 21, 22, 65533, 5] contains duplicate values
    spec_location: Global Elements

Contains two 65530.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...
[Uploading IDM-10.1.txt…]()

VaishaliAvhale, 2024-12-02:
Hi @ethan-jy,  

I have tested this on the latest commit `[commit 6d33b7d0ccafafe0e6040a5fc29a194ac0ffe197 (HEAD -> release/v1.3)]` of the `release/v1.3` branch, and I did not encounter this issue.  

I noticed that you have referenced this branch in this [ https://github.com/espressif/esp-matter/issues/1177] . Could you please confirm the branch you are using?

dhrishi, 2024-12-23:
@ethan-jy Please provide the requested details. Or please close the issue if it works

dhrishi, 2025-03-03:
@ethan-jy Closing this now. Please re-open in case of any problems

closed 2025-03-03 by dhrishi