espressif/esp-matter #1179: Which XML to use for harness test other than main device type

bilalmalik76, 2024-11-30:
I am trying to run harness test for a device which supports 6 on/off endpoints and 6 level control end points. Which XML I will need other than on/off and level control XMLs?

bilalmalik76, 2024-12-02:
Can you please check if these are all mandatory clusters for Root node or any else cluster also required:
[image]

VaishaliAvhale, 2024-12-02:
The selected file is correct please add this three as well
1. Access Control Enforcement
2. Group Communication
3. Group Key Management

bilalmalik76, 2024-12-02:
@VaishaliAvhale why pics generator did not generate above three XMLs, I used esp-matter light example also.

VaishaliAvhale, 2024-12-09:
The PICS Generator script has been written by the CSA team and is currently under development. Once it is ready, we will be able to generate XML files for all the clusters present on the device.

dhrishi, 2025-01-25:
@bilalahmaddev Is this solved? Can you please close this?

closed 2025-01-25 by bilalahmaddev