espressif/esp-matter #1180: Using AddResponse() to send a single integer value back to the Matter controller

pavel808, 2024-11-30:
I have implemented a customized cluster server on one of my end devices. I have commands which should be sent from a Matter controller to read value from sensors and send them back to the controller. I am still struggling to implement this properly. I can't find an example of doing this as a reference. 

This is my callback which gets invoked when the command gets received on the end device : 

```
static esp_err_t custom_command_callback(const ConcreteCommandPath &command_path, TLVReader &tlv_data,
                                                           void *opaque_ptr)
{
    // For simplicity, let's just say that I want to send back a single integer value back to the controller 
    int my_value = 5; 
    // I guess that I should be able to send this value as follows by encoding it somehow ? I'm really not sure:  
    // commandObj->AddResponse(commandPath, response);
}
```

On the Matter controller side this is how I send the command below. I would like to somehow be able to read the value back via my `command_send_success_cb() function. Is this even possible? Thanks in advance. 

```
esp_matter::lock::chip_stack_lock(portMAX_DELAY);

esp_matter::controller::cluster_command* command = new esp_matter::controller::cluster_command(nodeID, endpoint_id, custom_cluster_id, custom_command_id, commandData, chip::NullOptional, command_send_success_cb, command_send_error_cb);

esp_err_t err = command->send_command();

if (err != ESP_OK)
{
    ESP_LOGE(TAG, "ERROR! Unable to send cluster command : %d", err);
}

esp_matter::lock::chip_stack_unlock();
```

sayondeep, 2024-12-03:
@pavel808 
If I understand correctly. In simple terms, Once the end device receives the command it should send 5 as response back to the controller.

It would really help me with development if you can share the code for the controller and end-device to test.
For the starters, [this](https://github.com/project-chip/connectedhomeip/blob/2bc325178516defcf6706e5df793af1b55645fe1/src/app/CommandHandler.h#L188) is AddResponseData. This is what you are looking for.

pavel808, 2024-12-03:
> @pavel808 If I understand correctly. In simple terms, Once the end device receives the command it should send 5 as response back to the controller.
>
> It would really help me with development if you [...]

Hi @sayondeep Thanks for your reply. I'm just using an integer with hard-coded value 5 at the moment for simple testing. Exactly, from once the end-device receives the command it should send this integer value back to the controller. 

In reality, it will be reading a GPIO input when the command is received and send back the GPIO value read as an integer to the controller.

pavel808, 2024-12-04:
@sayondeep 

I've created a new issue with my code describing the probelm i'm facing :  https://github.com/espressif/esp-matter/issues/1192 

Maybe I should close this issue here.

closed 2024-12-06 by dhrishi