espressif/esp-matter #1181: How to change matter device type?

chuckrui, 2024-12-02:
Project: esp-matter/connecthomeip/connecthomeip/examples/lighting-app/esp32
when i commissioned it to apple home app,it show the device was "Dimmable light",but i want to change it to "On/Off light".
Whether I modify #CONFIG_DEVICE_TYPE from project build/sdkconfig.h or from idf.py menuconfig option Component->CHIP Device Layer->Device Identification Options->Default Device Type, none of which works.
I want to know that this value is required to be of type int, is it a decimal number corresponding to the Matter Device ID?
(0x0100-On/Off Light, 0x0101-Dimmable Light).

jadhavrohit924, 2024-12-02:
@chuckrui connectedhomeip's lighting app is a Zap-based application configured to Dimmable Light in the .zap file. You can find that at examples/lighting-app/lighting-common/lighting-app.zap. If you want to change the Dimmable Light to On/Off Light, you have to open the lighting-app.zap file using the zap tool, edit it, save it, rerun zap generation, and build your application.
```
1. cd connectedhomeip/connectedhomeip
2. ./scripts/tools/zap/run_zaptool.sh examples/lighting-app/lighting-common/lighting-app.zap
3. ./scripts/tools/zap/generate.py examples/lighting-app/lighting-common/lighting-app.zap
4. idf.py build
```

chuckrui, 2024-12-02:
@jadhavrohit924 
@jadhavrohit924
Yes, I modified it according to your instructions, and I ran three tests, respectively

1.Matter On/Off Light(0x0100)
[image]
[image]

2.Matter Dimmable Light(0x0101)
[image]
[image]

3.Matter Temperture Light(0x010C)
[image]
[image]

I found that Dimmable Light and Temperture Light are the same in the home app. Only On/Off Light is different.
Normally, On/Off Light can only be turned on and off, and Dimmable Light can also adjust the brightness besides turning on and off. Temperture Light can be turned on and off to adjust the brightness and color temperature, is there any problem?

jadhavrohit924, 2024-12-02:
No, there isn't any problem. This is expected behavior.

chuckrui, 2024-12-02:
Matter dimmable light Is color temperature adjustment normal for this device type? Because I set the same device type on the nordic52840 using the zap tool, the home app shows only dimming and on/off and no color temperature. However, according to your method, the zap tool is also used to change the device type to Matter dimmable light (0x0101), but the color temperature adjustment in the home app!
Because we already have a product that has been certified, and the device type is 0x0101, that is, Matter Dimmable Light. When the network is connected to the home app, no color temperature adjustment is displayed, only on/off and level control are displayed.
[image]

jadhavrohit924, 2024-12-02:
I don’t see color controls for dimmable light. I'm using iPadOs 18.0 What is yours?
[image]

jadhavrohit924, 2024-12-04:
> color control appears in both Iphone 18.1.1 and iPadOs 18.1 test "Matter Dimmabl Light"(esp32c3)

I have tested with iPadOS 18.1.1 and color control doesn’t appear.

>idf.py menuconfig option Component->CHIP Device Layer->Device Identification Options->Default Device Type
and the “CHIP_DEVICE_CONFIG_DEVICE_TYPE” value in Project build folder “”CHIPDeviceConfig.h”

Both are 0.

chuckrui, 2024-12-04:
I've solved the problem!
Should have nothing to do with OS and IOS system!
After the Device is changed to Matter Dimmable Light using the zap tool, the color control will appear in the homekit because the color control cluster is not disabled. So color control appears even if the device type is Matter Dimmable Light.

closed 2024-12-04 by chuckrui