espressif/esp-matter #1182: CommissioningWindow

MaplestoryAlen, 2024-12-02:
When the device is Commissioning, the WIFI signal is very poor. When the device is in "CHIP minimal mDNS started advertising.", the mobile WIFI is disconnected. At this moment, the device cannot connect. After closing the Commissioning window, the error returned is 3. Is there any way to close and then reopen the Commissioning window when the device opens it

code:
	  chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
	   commissionMgr.CloseCommissioningWindow();	   
	   chip::Server::GetInstance().GetFabricTable().DeleteAllFabrics();
          constexpr auto k_timeout_seconds = 300;

	  constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(k_timeout_seconds);
	  if (!commissionMgr.IsCommissioningWindowOpen())
	 {	      
		CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds,
					   chip::CommissioningWindowAdvertisement::kAllSupported);
		if (err != CHIP_NO_ERROR)
		{
		   global_info.PairedStatus = QRTimeout;
		   USER_LOGE(TAG, "Failed to open commissioning window, err:%" CHIP_ERROR_FORMAT, err.Format());
		}
	}
	else
	{
	    USER_LOGW(TAG, "Already open Commissioning window !");		
	}

log:

I (86074) chip[SVR]: Closing pairing window
I (86075) chip[DIS]: Updating services using commissioning mode 0
I (86081) chip[DIS]: CHIP minimal mDNS started advertising.
I (86092) chip[DIS]: Advertise operational node F87413946EAAAB83-090A1F7CC0E26000
I (86095) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: F87413946EAAAB83-090A1F7CC0E26000.
I (86109) chip[DIS]: mDNS service published: _matter._tcp
E (86112) chip[FP]: Reverting pending fabric data for fabric 0x1
E (86119) chip[FP]: Warning: metadata not found during delete of fabric 0x1
I (86129) chip[FP]: Fabric (0x1) deleted.
I (86131) chip[ZCL]: OpCreds: Fabric index 0x1 was removed
I (86138) chip[DIS]: Updating services using commissioning mode 0
I (86145) chip[DIS]: CHIP minimal mDNS started advertising.
I (86171) chip[TS]: Pending Last Known Good Time: 2024-12-02T10:22:45
I (86172) chip[TS]: Previous Last Known Good Time: 2023-10-14T01:16:48
I (86175) chip[TS]: Reverted Last Known Good Time to previous value
I (86182) chip[DIS]: Updating services using commissioning mode 0
I (86189) chip[DIS]: CHIP minimal mDNS started advertising.
E (86195) app_driver: Failed to open commissioning window, err:3

shubhamdp, 2024-12-03:
I think when you open the commissioning window, and starts pairing from another ecosystem app, it will arm the fail safe timer.

Either you should wait till that time much time before opening commissioning window, or disarm the fail safe timer.

I think the error is from here https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.cpp#L252 which checks if the fail safe is armed or not.

You can try disarming the fail safe timer along with closing the commissioning window. Please check https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.cpp#L575 for reference. `ExpireFailSafeIfArmed()` is a private API, so cannot be used directly.

MaplestoryAlen, 2024-12-04:
> I think when you open the commissioning window, and starts pairing from another ecosystem app, it will arm the fail safe timer.
> 
> Either you should wait till that much time before opening commissioning window, or disarm the fail safe timer.
> 
> I think the error is from here https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.cpp#L252 which checks if the fail safe is armed or not.
> 
> You can try disarming the fail safe timer along with closing the commissioning window. Please check https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.cpp#L575 for reference. `ExpireFailSafeIfArmed()` is a private API, so cannot be used directly.

Thank you for your answer. After closing "Failed Safe", you can close the Commissioning Window

        auto & failSafeContext = Server::GetInstance().GetFailSafeContext();
	   if (failSafeContext.IsFailSafeArmed()) 
	   {
		  failSafeContext.ForceFailSafeTimerExpiry();
	   }

closed 2024-12-04 by MaplestoryAlen