espressif/esp-matter #1183: Matter test hardness: Test cases failing on show test logs

bilalmalik76, 2024-12-02:
Matter test hardness: Test cases failing on show test logs

Environment

ESP-Matter Commit Id:origin/release/v1.3

ESP-IDF Commit Id:5.2.2

SoC: ESP32-S3

Device Logs (Please attach the log file):
[UI_Test_Run_2024_12_02_16_18_32.log](https://github.com/user-attachments/files/17977253/UI_Test_Run_2024_12_02_16_18_32.log)

Host Machine OS:

Commissioner's logs if present:
[image]

VaishaliAvhale, 2024-12-02:
Could you please share the Test Harness version or the commit ID you are currently using?

bilalmalik76, 2024-12-02:
@VaishaliAvhale I am using this branch v2.10.2+spring2024

bilalmalik76, 2024-12-03:
@VaishaliAvhale any update on this please?

VaishaliAvhale, 2024-12-03:
I'm looking into it and will get back to you soon.

bilalmalik76, 2024-12-03:
@VaishaliAvhale I also tried this but still no success:
[image]

bilalmalik76, 2024-12-05:
I am able to resolve this issue by disabling logs and executing onboarding tests before running Python automated tests.

closed 2024-12-05 by bilalmalik76