espressif/esp-matter #1186: How to create a new esp matter device

GayathriHommi, 2024-12-03:
**Describe the question/query that you have**
I want to use esp32h2, a temperature/humidity sensor and apple homepod as the border router. I created a sample project in ESP-IDF for I2C and added code for getting the temp/humidity values but now I want to integrate matter and I can't understand which platform should I use to develop matter project? Also how to generate QR code for a  matter project. Can we generated the code by ourselves or should we get support from CSA or espressif?

dhrishi, 2024-12-03:
@GayathriHommi You can refer the [sensors](https://github.com/espressif/esp-matter/tree/main/examples/sensors#matter-sensors) example that creates a temperature and a humidity sensor on two separate endpoints. Then just integrate the dirver related changes as per your hardware and you should be good to go.

For development, test data is used in the examples for the QR code along with test certs (Device Attestation Certificate). You can see/scan the QR code from [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning).

For production, please check the [Production Considerations section](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/production.html#production-considerations).

GayathriHommi, 2024-12-03:
@dhrishi, I have disabled WiFi chip station, build and  flashed the sensor firmware into esp32h2.
[image]
I scanned 
[image]
this QR code but connection is getting failed

shubhamdp, 2024-12-03:
Can you please try with attached [sdkconfig.defaults.esp32h2.txt](https://github.com/user-attachments/files/17994063/sdkconfig.defaults.esp32h2.txt), please rename this to `sdkconfig.defaults.esp32h2` and place it in `/path/to/esp-matter/examples/sensors` directory, and build the app

```
idf.py set-target esp32h2 build
```

---
I tested this with esp-thread-border-router and an H2 devkit and commissioning works.

GayathriHommi, 2024-12-03:
Thank you for the support, It is connecting now

closed 2024-12-03 by GayathriHommi

shubhamdp, 2024-12-04:
@GayathriHommi we have added the above `sdkconfig.defaults.h2` to the [sensor app](https://github.com/espressif/esp-matter/tree/main/examples/sensors) in 70eb531.

Please pull the latest changes and you won't need to maintain the separate copy.

GayathriHommi, 2024-12-05:
@shubhamdp Can I open this esp-matter/sensor code in ESP IDF? Previously I'm using visual Studio and I can't enter to debug mode (while building errors are prompted). I can only modify, build and flash in the vs terminal.

shubhamdp, 2024-12-05:
@GayathriHommi please check [vscode-esp-idf-extension](https://github.com/espressif/vscode-esp-idf-extension?tab=readme-ov-file#all-available-commands), it supports setting up esp-matter.