espressif/esp-matter #1187: PICS files of Multiple endpoints in Matter harness test

bilalmalik76, 2024-12-03:
If my device has multiple identical endpoints, such as 3 On/Off and 3 Level Control endpoints, do I need to upload the PICS XML for all of them during the Matter harness test? If yes, why is it required, considering that all identical endpoints will have the same PICS files?

wqx6, 2024-12-11:
If the data model on the three application endpoints are the same, you could provide PICS files for one application endpoint and notice the ATL that all the application endpoints are the same. 
But it is better to provide PICS files in the following file directory structure
```
-- PICS ---- Base.xml
         |
         --- endpoint0 ---- cluster-pics-files on endpoint 0
         | .....
         --- endpointX ---- cluster-pics-files on endpoint X
```

bilalmalik76, 2024-12-11:
Great! I have done this already correctly then by assuming this. :)

dhrishi, 2024-12-24:
Closing this now

closed 2024-12-24 by dhrishi