espressif/esp-matter #1188: ESP-Matter and ESP-NOW

sdrasti, 2024-12-03:
I have a Matter DoorLock app running on an ESP32S3 and have also activated ESP-NOW, which sends about 20 bytes every 500ms to another ESP32, that currently just logs the incoming data.

The ESP-NOW data transfer still works fine when the Matter core is started, but fails completely as soon as the device is commissioned via BLE/Wifi.

I tried to get it working again by retrieving the wifi channel after the commissioning finished with `esp_wifi_get_config` and passing it to `esp_now_add_peer` plus setting `esp_wifi_set_ps(WIFI_PS_NONE)`, but that doesn't bring the ESP-NOW connection back to life.

The moment I remove the device from the network, ESP-NOW works again without any problems.

Does the Wifi handling in ESP-Matter present any potential obstacles or critical issues that could prevent the simultaneous use of ESP-NOW on the same chip?

bubblesnake, 2024-12-04:
@sdrasti , have you changed the responder's own operating channel to be the same as the initiator's new channel?
https://github.com/espressif/esp-idf/blob/030c9957dbc5c5a627cf1bb4b2b856f1b6c1343c/examples/wifi/espnow/main/espnow_example_main.c#L54

sdrasti, 2024-12-04:
@bubblesnake 

Thanks! With updating the wifi channel on the responder side that has been changed through the commissioning process ESP-NOW works without any issues again.

But what is the channel setting in the esp_now_add_peer config good for then? I would expect it to set the channel of the responder appropriately. Otherwise, how can one change the channel dynamically in the responder in this scenario?

closed 2024-12-05 by sdrasti