espressif/esp-matter #1189: Adding Sonoff sensor support for Matter Zigbee Bridge example

CreLab, 2024-12-03:
Is it possible to add Sonoff sensors to the Matter Zigbee Bridge, especially temperature and humidity sensors?

I have installed the Matter Zigbee Bridge on my Espressif Zigbee Gateway board and successfully added it to the Home Assistant. But I can't find a solution to add a Sonoff sensor to the Matter Zigbee Bridge.

Please give me more information on how to add sensors to the Matter Zigbee Bridge. And if it is not supported, how can I add this feature to the Matter Zigbee Bridge source code example?

wqx6, 2024-12-18:
Hi, does the sonoff sensor use install code to join a ZigBee network?
If yes, you can call `esp_zb_zdo_permit_joining_req()` and `esp_zb_secur_ic_add()` to allow the sensor end device joining the network. And you should also add the device type of the sensor to the [`create_bridge_devices()`](https://github.com/espressif/esp-matter/blob/main/examples/bridge_apps/zigbee_bridge/main/app_main.cpp#L82). so that the dynamic endpoint can be added to the matter data model after the sensor joins the zigbee network.
If no, you need to find the way how the sonoff sensor joins a ZigBee network, and add it to the Zigbee network.

dhrishi, 2025-03-03:
@CreLab Did you get a chance to try this out? Can you close this issue if it works fine?

dhrishi, 2025-04-17:
@CreLab Please close the issue