espressif/esp-matter #1190: Attestation - Secure Cert, SendAttestationRequest failed

ethan-jy, 2024-12-04:
**Describe the bug**
Chip-tool debugging failed
command:
[image]
esp-matter-mfg-tool  -cn meanwell -v 0xFFF2 -p 0x8001 --pai \
   -k /home/meanwell/espressif/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c /home/meanwell/espressif/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd /home/meanwell/espressif/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
   --dac-in-secure-cert --target esp32h2 -n 2 --product-name meanwell --hw-ver-str "1.0" --vendor-name meanwell

xxx_esp_secure_cert.bin  download to 0xd000;
xxx-partition download to 0x3E0000;

DAC Provider set to Attestation - Secure Cert
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL = fctry = 0x3E0000

**Environment**
- ESP-Matter Commit Id:1.3
- ESP-IDF Commit Id:5.3.1
- SoC (eg: ESP32 or ESP32-C3):esp32h2
- Device Logs (Please attach the log file):ubuntu
[Uploading Chip-tool Log.txt…]()

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

ethan-jy, 2024-12-04:
[Chip-tool Log.txt](https://github.com/user-attachments/files/18001358/Chip-tool.Log.txt)
[SDK config.txt](https://github.com/user-attachments/files/18001359/SDK.config.txt)
Please review the recording

ethan-jy, 2024-12-04:
[image]
[image]

ethan-jy, 2024-12-04:
What is this problem and how should it be handled?

shripad621git, 2024-12-06:
@ethan-jy , thanks for reporting the issue. We are looking into it. We will get back with the solution soon.

shripad621git, 2024-12-10:
@ethan-jy , we have identified the issue. The esp-matter-mfg-tool does not burn the private key in the `efuse` in case of h2. For more information on `efuse` and `esp_secure_cert_partition` in case of `esp32h2`, you can refer [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/platforms/esp32/secure_cert_partition.md).
We are in process of adding the support and testing it out. You will get to test the functionality soon.

shripad621git, 2025-01-08:
@ethan-jy , the esp-matter-mfg-tool is now updated with the efuse related code [here](https://github.com/espressif/esp-matter-tools/commit/b5e40150c2b18619c233101863d9a06789c9f5c5).
Please  install the latest `esp-matter-mfg-tool `and run the below command.
If you want 2 esp_secure_cert partitions for esp32h2, please run the below command on two different devkits separately as the efuse needs to be burnt separately for each chip and is a irreversible operation.
Enable appropriate options in the menuconfig as done before.

```
esp-matter-mfg-tool -cn meanwell -v 0xFFF2 -p 0x8001 --pai \
-k /path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c /path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd /path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--dac-in-secure-cert --target esp32h2 --product-name meanwell --hw-ver-str "1.0" --vendor-name meanwell --ds-peripheral --efuse-key-id 0 --port /dev/ttyUSB0

```

abu-matterize, 2025-01-18:
I've similar issue. The target device is `esp32` , I've the DAC cert, key and PAI in place and generated `**_esp_secure_cert.bin` and `**-parition.bin` with `esp-matter-mfg-tool` and flashed to their respective addresses.

`esp-matter-mfg-tool --target esp32 -cf 0 -dm 1 --pai -c Downloads/Output/certificates/some_PAI.pem --dac-cert Downloads/Output/certificates/some.pem --dac-key Downloads/Output/certificates/some_key.pem  -v 0xFFF1 -p 0x8001 --dac-in-secure-cert --mfg-date 2025-01-18 --serial-num 12345`

The device commissioning fails at verifying opcred.

Device log:
```sh
I (56198) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (56198) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (56218) chip[EM]: >>> [E:41793r S:55050 M:88048318] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (56218) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x0000003E
I (56238) chip[ZCL]: OpCreds: Received an AttestationRequest command
E (56248) chip[DMG]: Endpoint=0 Cluster=0x0000_003E Command=0x0000_0000 status 0x01 (no additional context)
E (56248) chip[ZCL]: OpCreds: Failed AttestationRequest request with IM error 0x01 (err = 201)
I (56258) chip[EM]: <<< [E:41793r S:55050 M:95743195] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:09 (IM:InvokeCommandResponse)
I (56278) NimBLE: GATT procedure initiated: indicate;
I (56278) NimBLE: att_handle=18

I (56348) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (56348) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (56368) chip[EM]: >>> [E:41794r S:55050 M:88048319] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (56368) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x00000030
I (56388) chip[FS]: GeneralCommissioning: Received ArmFailSafe (0s)
I (56398) chip[FS]: Fail-safe timer expired
I (56398) chip[EM]: <<< [E:41794r S:55050 M:95743196] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:09 (IM:InvokeCommandResponse)
I (56408) NimBLE: GATT procedure initiated: indicate;
I (56418) NimBLE: att_handle=18

E (56418) chip[SVR]: Failsafe timer expired
I (56418) chip[SC]: SecureSession[0x3ffc6548, LSID:55050]: State change 'kActive' --> 'kPendingEviction'
E (56428) chip[SVR]: Commissioning failed (attempt 1): 32
I (56448) chip[BLE]: Releasing end point's BLE connection back to application.
I (56448) chip[DL]: Closing BLE GATT connection (con 0)
I (56458) NimBLE: GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
```

I'm using the default partition layout used in [light](https://github.com/espressif/esp-matter/blob/main/examples/light/partitions.csv)

shripad621git, 2025-04-21:
@abu-matterize , can you please mention if there was a specific reason to enable `CONFIG_ENABLE_SET_CERT_DECLARATION_API=y ` in the firmware.
As enabling [this](https://github.com/espressif/connectedhomeip/blob/326cabf99c0cd2d8e9099ea0a5cc849c8e28dda7/config/esp32/components/chip/Kconfig#L854) , `you need to set the cetrtification-declaration through firmware`.
Also, during commissioning, did you provide the correct path to the `paa certificate directory holding paa certificate using --paa-trust-store-path` option?

For e.g: 
```
./chip-tool pairing ble-wifi [node-id] [ssid] [password] [passcode][discriminator] --paa-trust-store-path /path/to/paa-cert-dir
```