espressif/esp-matter #1192: Sending data from end device back to controller using cluster command issue

pavel808, 2024-12-04:
I have a customized cluster server on one of my C6 end devices. 

Also I have a Matter controller application on another S3 device which should be able to read data from the end device when it issues a cluster command to it. Commissioning etc. all works fine. There is just an issue when trying to read the value back in the Matter controller as below. 

Here is my code for each device which deals with the cluster command. The end device encodes the data into a `ForcedSizeBuffer` and tries to send the data using `AddResponseData()` . All seems to work fine, but this fails in the Matter Controller application as` TagNumFromTag(tag) `is 0.

Does the below logic make sense for each device? In other words, should this work, or why is it failing to decode the response on the Matter Controller side? Thanks in advance.  

This is what is implemented on my end device to handle the command. I'm just hard-coding a `test_value` of 55 for now which should be sent to the controller when the command is received. 

```
class ForcedSizeBuffer : public chip::app::DataModel::EncodableToTLV
{
public:
    ForcedSizeBuffer(uint32_t size, uint32_t val )
    {
        if (mBuffer.Alloc(size))
        {
            // No significance with using 0x12, just using a value.
            memset(mBuffer.Get(), val, size);
        }
    }

    // No significance with using 0x12 as the CommandId, just using a value.
    static constexpr chip::CommandId GetCommandId() { return 0x12; }
    CHIP_ERROR EncodeTo(chip::TLV::TLVWriter & aWriter, chip::TLV::Tag aTag) const override
    {
        VerifyOrReturnError(mBuffer, CHIP_ERROR_NO_MEMORY);

        chip::TLV::TLVType outerContainerType;
        ReturnErrorOnFailure(aWriter.StartContainer(aTag, chip::TLV::kTLVType_Structure, outerContainerType));
        ReturnErrorOnFailure(chip::app::DataModel::Encode(aWriter, chip::TLV::ContextTag(1), chip::ByteSpan(mBuffer.Get(), mBuffer.AllocatedSize())));
        return aWriter.EndContainer(outerContainerType);
    }

private:
    chip::Platform::ScopedMemoryBufferWithSize<uint8_t> mBuffer;
};

static esp_err_t read_status_callback(const ConcreteCommandPath &command_path, TLVReader &tlv_data,
                                                           void *opaque_ptr)
{
	/* Get ids */
	uint16_t endpoint_id = command_path.mEndpointId;
	uint32_t cluster_id = command_path.mClusterId;
	uint32_t command_id = command_path.mCommandId;

	/* Return if this is not the correct cluster and command*/
	if (endpoint_id != light_endpoint_id || cluster_id != ir_cluster_id || command_id != read_usb_status_id)
	{
            ESP_LOGE(TAG, "ERROR! read_usb_status_callback has incorrect ids for endpoint, cluster or command");
	    return ESP_FAIL;
	}

	ESP_LOGI(TAG, "About to read the USB voltage...");

	chip::app::CommandHandler *command_obj = (chip::app::CommandHandler *)opaque_ptr;
	if (!command_obj)
	{
	    ESP_LOGE(TAG, "Command Object cannot be NULL");
	    return ESP_FAIL;
	}

	// Test value initially
	int test_value = 55;

	ForcedSizeBuffer responseData(sizeof(uint32_t), test_value);

	ESP_LOGI(TAG, "About to send response data back to controller for endpoint : 0x%x , cluster ID 0x%lx and command ID 0x%lx", endpoint_id, cluster_id, command_id);

	chip::ChipError chip_err = command_obj->AddResponseData(command_path, (chip::CommandId)command_id, responseData);
	if (chip_err != CHIP_NO_ERROR)
	{
            ESP_LOGE(TAG, "Unable to send response data from read_usb_status_callback err:%" CHIP_ERROR_FORMAT, chip_err.Format());
	    return ESP_FAIL;
	}

	ESP_LOGI(TAG, "Sending success status back to Server for endpoint : 0x%x , cluster ID 0x%lx and command ID 0x%lx", endpoint_id, cluster_id, command_id);

	command_obj->AddStatus(command_path, chip::Protocols::InteractionModel::Status::Success);

        return ESP_OK;
}

// Creating the command in the main() function
command_t *read_status_command = command::create(ir_cluster, read_status_id, COMMAND_FLAG_ACCEPTED | COMMAND_FLAG_CUSTOM, read_status_callback);
if (read_usb_status_command == NULL)
{
        ESP_LOGE(TAG, "ERROR! Unable to create read_usb_status_command : 0x%lx for endpoint 0x%x ", read_usb_status_id, endpoint_id);
        return ESP_FAIL;
}
```

This is what is implemented on the Matter Controller application. Commands get sent and the `command_send_success_cb() `gets invoked. However, this fails on `if (0 != TagNumFromTag(tag))`. Even if I remove this check, the decoding fails in the next step. 

```
void command_send_success_cb(void *ctx, const ConcreteCommandPath &command_path, const chip::app::StatusIB &status,
                                          TLVReader *response_data)
{
    ESP_LOGI(TAGD, "Cluster command success cb");

    uint16_t endpoint_id = command_path.mEndpointId;
    uint32_t cluster_id = command_path.mClusterId;
    uint32_t command_id = command_path.mCommandId;

    uint32_t data = 0;

    ESP_LOGI(TAGD, "endpoint_id : %lX", endpoint_id );
    ESP_LOGI(TAGD, "Command ID : %llX", command_id);

    if (response_data == NULL)
    {
    	ESP_LOGI(TAGD, "No response data to process");
    	return;
    }

    ESP_LOGI(TAGD, "tlv data type  %d", response_data->GetType());

    if (chip::TLV::kTLVType_Structure != response_data->GetType())
    {
	ESP_LOGE(TAGD, "response_data->GetType() ERROR");
	return;
    }

    chip::TLV::TLVType mOuter;
    if (CHIP_NO_ERROR != response_data->EnterContainer(mOuter))
    {
	ESP_LOGE(TAGD, "mOuter ERROR");
	return;
    }

    if (CHIP_NO_ERROR != response_data->Next())
    {
	ESP_LOGE(TAGD, "response_data->Next() ERROR");
	return;
    }

    chip::TLV::Tag tag = response_data->GetTag();
    if (!IsContextTag(tag))
    {
	ESP_LOGE(TAGD, "response_data->GetTag() ERROR");
	return;
    }

    // THIS FAILS !!
    // check tag number, since this has only one argument, checking against 0
    if (0 != TagNumFromTag(tag))
    {
	ESP_LOGE(TAGD, "TagNumFromTag(tag) ERROR");
	return;
    }

    CHIP_ERROR chip_err = chip::app::DataModel::Decode(*response_data, data);
    if (chip_err != CHIP_NO_ERROR)
    {
	ESP_LOGI(TAGD, "ERROR! Unable to get TLV Response Data");
	return;
    }

    ESP_LOGI(TAGD, "Data received from device: %llu", data);
}

// From the main function

esp_matter::lock::chip_stack_lock(portMAX_DELAY);

cluster_command* command = new cluster_command(nodeID, endpoint_id, ir_cluster_id, read_status_id, NULL, chip::NullOptional, command_send_success_cb, command_send_error_cb);

err = command->send_command();

if (err != ESP_OK)
{
	ESP_LOGI(TAGD, "ERROR! Unable to send cluster command to read USB status : %d", err);
}

esp_matter::lock::chip_stack_unlock();
```

wqx6, 2024-12-05:
The check should be `0 != static_cast<uint8_t>(TagNumFromTag(tag))`. And you don't need to add the status on success. Add response is enough for success case. The status can be added  for failure case.

pavel808, 2024-12-05:
> The check should be `0 != static_cast<uint8_t>(TagNumFromTag(tag))`. And you don't need to add the status on success. Add response is enough for success case. The status can be added for failure case.

@wqx6 I've made that change with `0 != static_cast<uint8_t>(TagNumFromTag(tag))`, and also removed the AddStatus call from the device. Unfortunately it still fails for me with `TagNumFromTag(tag) ERROR` . The value of this is 1 and not the expected 0. 

On the device side, am I encoding the value correctly? I'm not sure why it fails to decode on the controller end. Thanks.

sayondeep, 2024-12-05:
@pavel808 

Can you please try with the follwing code changes:
On the end device custom cluster code 
```
class ForcedSizeBuffer : public chip::app::DataModel::EncodableToTLV
{
public:
    ForcedSizeBuffer(uint32_t val )
    {
        this->data = val;
    }

    // No significance with using 0x12 as the CommandId, just using a value.
    static constexpr chip::CommandId GetCommandId() { return 0x12; }
    CHIP_ERROR EncodeTo(chip::TLV::TLVWriter & aWriter, chip::TLV::Tag aTag) const override
    {
        chip::TLV::TLVType outerContainerType;
        ReturnErrorOnFailure(aWriter.StartContainer(aTag, chip::TLV::kTLVType_Structure, outerContainerType));
        CHIP_ERROR err = chip::app::DataModel::Encode(aWriter, chip::TLV::ContextTag(1), data);
        if(err != CHIP_NO_ERROR)
            ESP_LOGE(TAG,"Error encoding value");
        return aWriter.EndContainer(outerContainerType);
    }

private:
    uint32_t data;
};
```

```
static esp_err_t read_status_callback(const ConcreteCommandPath &command_path, TLVReader &tlv_data,
                                                           void *opaque_ptr)
{
	/* Get ids */
	uint16_t endpoint_id = command_path.mEndpointId;
	uint32_t cluster_id = command_path.mClusterId;
	uint32_t command_id = command_path.mCommandId;

	/* Return if this is not the correct cluster and command*/
	if (endpoint_id != cluster::rainmaker::endpoint_id || cluster_id != cluster::rainmaker::Id || command_id != cluster::rainmaker::command::read_status::Id)
	{
            ESP_LOGE(TAG, "ERROR! read_usb_status_callback has incorrect ids for endpoint, cluster or command");
	    return ESP_FAIL;
	}

	ESP_LOGI(TAG, "About to read the USB voltage...");

	chip::app::CommandHandler *command_obj = (chip::app::CommandHandler *)opaque_ptr;
	if (!command_obj)
	{
	    ESP_LOGE(TAG, "Command Object cannot be NULL");
	    return ESP_FAIL;
	}

	// Test value initially
	uint32_t test_value = 55;

    ForcedSizeBuffer responseData(test_value);

	ESP_LOGI(TAG, "About to send response data back to controller for endpoint : 0x%x , cluster ID 0x%lx and command ID 0x%lx", endpoint_id, cluster_id, command_id);

	chip::ChipError chip_err = command_obj->AddResponseData(command_path, (chip::CommandId)command_id, responseData);
	if (chip_err != CHIP_NO_ERROR)
	{
            ESP_LOGE(TAG, "Unable to send response data from read_usb_status_callback err:%" CHIP_ERROR_FORMAT, chip_err.Format());
	    return ESP_FAIL;
	}

	ESP_LOGI(TAG, "Sending success status back to Server for endpoint : 0x%x , cluster ID 0x%lx and command ID 0x%lx", endpoint_id, cluster_id, command_id);

	command_obj->AddStatus(command_path, chip::Protocols::InteractionModel::Status::Success);

        return ESP_OK;
}
```

On controller side:

```
static void command_send_success_cb(void *ctx, const ConcreteCommandPath &command_path, const chip::app::StatusIB &status,
                                          TLVReader *response_data)
{
    ESP_LOGI(TAG, "Cluster command success cb");

    uint16_t endpoint_id = command_path.mEndpointId;
    uint32_t cluster_id = command_path.mClusterId;
    uint32_t command_id = command_path.mCommandId;

    uint32_t data = 0;

    ESP_LOGI(TAG, "endpoint_id : %lX", endpoint_id );
    ESP_LOGI(TAG, "Command ID : %llX", command_id);

    if (response_data == NULL)
    {
    	ESP_LOGI(TAG, "No response data to process");
    	return;
    }

    ESP_LOGI(TAG, "tlv data type  %d", response_data->GetType());

    if (chip::TLV::kTLVType_Structure != response_data->GetType())
    {
	ESP_LOGE(TAG, "response_data->GetType() ERROR");
	return;
    }

    chip::TLV::TLVType mOuter;
    if (CHIP_NO_ERROR != response_data->EnterContainer(mOuter))
    {
	ESP_LOGE(TAG, "mOuter ERROR");
	return;
    }

    if (CHIP_NO_ERROR != response_data->Next())
    {
	ESP_LOGE(TAG, "response_data->Next() ERROR");
	return;
    }

    chip::TLV::Tag tag = response_data->GetTag();
    if (!IsContextTag(tag))
    {
	ESP_LOGE(TAG, "response_data->GetTag() ERROR");
	return;
    }

    ESP_LOGI(TAG, "tlv inside data type  %d", response_data->GetType());

    CHIP_ERROR chip_err = chip::app::DataModel::Decode(*response_data, data);
    if (chip_err != CHIP_NO_ERROR)
    {
	ESP_LOGI(TAG, "ERROR! Unable to get TLV Response Data");
	return;
    }

    ESP_LOGI(TAG, "Data received from device: %d", data);
}
```

pavel808, 2024-12-05:
@sayondeep Now it works. Thank you so much.

closed 2024-12-06 by dhrishi