espressif/esp-matter #1193: Tags and Docker Containers for Certified OpenThread Versions

jzwyssig, 2024-12-05:
In order to inherit the Thread certification, companies need to proof that the OpenThread library version they use matches the certified version of Espressif. For example, during the Thread 1.3 certification process, we had to manually check out a specific commit of the OpenThread library. As this specific commit was not referenced in any esp-idf commit, we had to checkout the submodule manually.

In the future, it would be very helpful if Espressif could:

- **Create tags** for certified OpenThread versions. The next certified release will be Thread 1.4, probably with esp-idf 5.3 or 5.4.
- **Provide Docker containers** for ESP-Matter and ESP-IDF that include the exact certified version of the OpenThread library.

This would require minimal effort on your end but would simplify the certification process for many companies. For example, when a company uses the espressif/esp-matter docker container that uses the certified Thread version.

Thank you for considering this request.

chshu, 2024-12-24:
@jzwyssig We have an official process to support our Thread customers in inheriting the platform's Thread certification. Please reach out to our team via [Espressif Technical Inquiries](https://www.espressif.com/en/contact-us/technical-inquiries), and we will provide the Thread certification ID along with the corresponding IDF release versions.

jzwyssig, 2025-01-09:
Thank you, @chshu, for your answer. We have already reached out to the Espressif team and received the commit hash of the OpenThread library used for the certification. However, I am suggesting that it would be great to have a separate tag for your next certified (probably Thread 1.4) version in ESP-IDF and ESP-Matter. A big plus would also be having a separate Docker container with the certified stack. This would help companies with Matter certification, as they could simply use the Docker container or check out the corresponding tag in ESP-Matter.

closed 2025-01-29 by jzwyssig