espressif/esp-matter #1194: OnOff restore state after reboot

stefano055415, 2024-12-05:
I have a problem with the initial state of the light after a reboot (it always goes back to off), this is the configuration

on_off_light::config_t light_config;
    light_config.on_off.on_off = false;
    light_config.on_off.lighting.start_up_on_off = nullptr;

these are the logs

I (4402) esp_matter_core: Dynamic endpoint 0 added
I (4402) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000003's Attribute 0x00000001 is 0 **********
I (4422) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (4432) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x0000FFFC is <invalid type: 0> **********
I (4442) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000005's Attribute 0x00000000 is 0 **********
I (4452) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (4462) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00004003 is 0 **********
I (4472) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 **********
I (4492) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 **********
I (4502) chip[ZCL]: Toggle ep1 on/off from state 1 to 0
I (4502) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (4532) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (4532) chip[ZCL]: Off completed. reset OnTime to  0
I (4542) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00004001 is 0 **********
I (4552) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000005's Attribute 0x00000003 is 0 **********

the problem occurs with the release/1.1, while with release/1.3 it works without problems

jadhavrohit924, 2024-12-06:
Have you erase-flash the device? PTAL https://github.com/espressif/esp-matter/issues/841#issuecomment-1957530500

jadhavrohit924, 2024-12-17:
@stefano055415 Any update on the issue?

stefano055415, 2024-12-17:
@jadhavrohit924 

To resolve after matter has started you must set this.

nullable<unsigned char> nullable_value;
esp_matter_attr_val_t startup_val = esp_matter_nullable_enum8(nullable_value);
attribute::update(endpoint_id, OnOff::Id, OnOff::Attributes::StartUpOnOff::Id, &startup_val);

Probably in some version the default value was set to 0

closed 2024-12-17 by stefano055415