espressif/esp-matter #1195: Temperature Control Cluster write attribute invalid type

algo-techai, 2024-12-06:
## Bug Description
When attempting to write the `TargetTemperature` attribute of the Temperature Control cluster for a refrigerator device with two temperature-controlled cabinets, the attribute is reported as an invalid type during the write operation.

- **Expected Behavior**: The `TargetTemperature` attribute value should be updated to the provided value (e.g., `260`) as per the CHIP specification.
- **Actual Behavior**: The write operation reports an "invalid type" for the `TargetTemperature` attribute, with a value type of `0` (unknown) in the logs.

---

## Steps to Reproduce
1. Create a Temperature Controlled Cabinet device with the following configuration:
    ```cpp
    temperature_controlled_cabinet::config_t fridge_temperature_controlled_cabinet_config;
    fridge_temperature_controlled_cabinet_config.temperature_control.temperature_number.temp_setpoint = 500;
    fridge_temperature_controlled_cabinet_config.temperature_control.temperature_number.min_temperature = 200;
    fridge_temperature_controlled_cabinet_config.temperature_control.temperature_number.max_temperature = 800;
    fridge_temperature_controlled_cabinet_config.temperature_control.temperature_level.selected_temp_level = 0;

    endpoint_t *fridge_cabinet_endpoint = temperature_controlled_cabinet::create(node, &fridge_temperature_controlled_cabinet_config, ENDPOINT_FLAG_NONE, NULL);
    esp_matter::cluster::temperature_measurement::config_t fridge_temperature_measurement_config;
    fridge_temperature_measurement_config.measured_value = 500;
    fridge_temperature_measurement_config.min_measured_value = 200;
    fridge_temperature_measurement_config.max_measured_value = 800;

    esp_matter::cluster::temperature_measurement::create(fridge_cabinet_endpoint, &fridge_temperature_measurement_config, CLUSTER_FLAG_SERVER);

    cluster_t *fridge_temperature_control_cluster = cluster::get(fridge_cabinet_endpoint, chip::app::Clusters::TemperatureControl::Id);
    esp_matter::cluster::temperature_control::feature::temperature_number::add(fridge_temperature_control_cluster, &(fridge_temperature_controlled_cabinet_config.temperature_control.temperature_number));
    ```

2. Use the `chip-tool` to send the `set-temperature` command:
    ```
    temperaturecontrol set-temperature 0x7283 0x2 --TargetTemperature 260
    ```

3. Observe the device logs for errors.

---

## Observed Logs
- Write operation on the `TargetTemperature` attribute reports the following:
    ```
    I (50456) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000056's Attribute 0x00000000 is <invalid type: 0> **********
    I (50476) app_main: Attribute update callback: type: 0, endpoint_id: 0x2, cluster_id: 0x56, attribute_id: 0x0, val_type: 0
    ```

- Despite the error, subsequent reads show that the value is correctly updated:
    ```
    I (50606) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000056's Attribute 0x00000000 is 260 **********
    ```

---

## Environment
- **ESP-Matter Commit ID**: 21c2051
- **ESP-IDF Commit ID**: c9763f62dd (idf version 5.2.3)
- **SoC**: ESP32-S3

- **Host Machine OS**: macOS 15.0
- **Host Machine Python Version**: 3.13.0

- **Commissioner App and Version**: `chip-tool`
- **Commissioner's Logs**: [Attach logs if available]
```
[1733509110.492] [77957:36220198] [TOO] Command: temperaturecontrol set-temperature 0x7283 0x2 --TargetTemperature 260 
[1733509110.493] [77957:37205455] [TOO] Sending command to node 0x7283
[1733509110.494] [77957:37205455] [CSM] FindOrEstablishSession: PeerId = [1:0000000000007283]
[1733509110.494] [77957:37205455] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1733509110.494] [77957:37205455] [DIS] Found an existing secure session to [1:0000000000007283]!
[1733509110.494] [77957:37205455] [DIS] OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1733509110.494] [77957:37205455] [TOO] Sending cluster (0x00000056) command (0x00000000) on endpoint 2
[... 43 lines cut ...]
[1733509110.947] [77957:37205455] [DMG]                                 },
[1733509110.947] [77957:37205455] [DMG] 
[1733509110.947] [77957:37205455] [DMG]                                 StatusIB =
[1733509110.947] [77957:37205455] [DMG]                                 {
[1733509110.947] [77957:37205455] [DMG]                                         status = 0x00 (SUCCESS),
[1733509110.947] [77957:37205455] [DMG]                                 },
[1733509110.948] [77957:37205455] [DMG] 
[1733509110.948] [77957:37205455] [DMG]                         },
[1733509110.948] [77957:37205455] [DMG] 
[1733509110.948] [77957:37205455] [DMG]                 },
[1733509110.948] [77957:37205455] [DMG] 
[1733509110.948] [77957:37205455] [DMG]         ],
[1733509110.948] [77957:37205455] [DMG] 
[1733509110.948] [77957:37205455] [DMG]         InteractionModelRevision = 11
[1733509110.948] [77957:37205455] [DMG] },
[1733509110.948] [77957:37205455] [DMG] Received Command Response Status for Endpoint=2 Cluster=0x0000_0056 Command=0x0000_0000 Status=0x0
[1733509110.948] [77957:37205455] [DMG] Received Command Response Status for Endpoint=2 Cluster=0x0000_0056 Command=0x0000_0000 Status=0x0
```
---

## Additional Details
- The `TargetTemperature` attribute is configured as part of the `temperature_number` feature of the Temperature Control cluster.
- The `step` value for the temperature was not added during the endpoint setup. Even if it was added, it gives the same bug

---

jadhavrohit924, 2024-12-10:
@algo-techai Thanks for the report. We will fix it.

algo-techai, 2024-12-16:
@jadhavrohit924  Any Updates?

closed 2024-12-17 by espressif-bot