espressif/esp-matter #1196: Matter Device Type recommendation

bilalmalik76, 2024-12-07:
We are working on a Matter device that supports multiple types of devices. It provides users the option to configure the device type through a configuration page. The Matter device can represent various types, such as on/off, dimmable, color control, etc. We need to dynamically add endpoints based on the device types selected by the user.

What are your recommendations for implementing this? Would using a bridge be a suitable option for such a Matter device?

shubhamdp, 2024-12-10:
Can you please check [all_device_types_app](https://github.com/espressif/esp-matter/tree/main/examples/all_device_types_app), it uses the cli for choosing the device type.
NOTE: This lets you have only one device per factory reset.

bilalmalik76, 2024-12-10:
@shubhamdp we want to support multiple device types and at a time device can support 4-5 device types.

shubhamdp, 2024-12-10:
Well in that case, you can start building on top of all_device_types_app. You may have to define the upper limit for fitting in all these in your selected SoC.

bilalmalik76, 2024-12-10:
@shubhamdp what about Matter bridge device?

chshu, 2024-12-18:
Regarding certification, for each device type supported by your product, you need to declare them in the PICS file and provide the lab with instructions on how to enable each device type and conduct the required testing.

Since a bridge is also a Matter device type, you must list all possible bridged device types and prepare the necessary non-Matter devices for testing.

You may refer to [certification](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html) for more instructions, and contact CSA for official answers.

jonsmirl, 2024-12-27:
Lighting is a special case, you are allowed to list the four of the device types on a single endpoint: on/off, dimmable, color temp, extended color. This is the simplest solution. You are not allowed to add more than those four. 

You are not allowed to combine device types from different devices onto a single endpoint.  Lighting is special because the four devices are a nested set. Each device type is a super set of the previous one.

It is possible to get fancier and dynamically change the list of device types... but none of the controller apps except Apple will recognize this. You need to update the Descriptor cluster to make Apple rescan.

A very advanced solution is to create a Matter bridge internal to the device with four endpoints for each possible lamp (on/off, dimmable, color temp, extended color) then only enable the correct one. That will work with all of the apps (Google, Amazon, Apple, Samsung) and display the right icons.

Unless you have a lot of free engineers available, I'd stick with the fist option.

bilalmalik76, 2024-12-27:
Hi @jonsmirl Thank you for your detailed response. 

I understand it is possible for Lighting case. In one of our device which is already in certification phase, we added 12 on/off and dimmable endpoints. Because this device has fixed number of predefined endpoints with each endpoint having fixed on/off or dimmable clusters, it works fine on all ecosystems. 

In the device we plan to develop, dynamic endpoints will be supported. However, once the user configures the device types through the user-programmable interface, they must stick to the selected device types unless they perform a Matter factory reset and recommission the device with different device types.

For instance, if the device is designed to support 12 endpoints, we will define fixed clusters for each endpoint. The user can then select the desired device type associated with each endpoint. Once the configuration is completed through the settings page, only the selected endpoints will be enabled and visible in the Matter app after commissioning.

@jonsmirl what do you think? Matter bridge is also something we are considering.

jonsmirl, 2024-12-27:
What you described will work. To do that dynamically and have it work on all of the apps, you have to make an internal bridge.

bilalmalik76, 2024-12-27:
@jonsmirl We are also thinking to go with internal bridge. what do you say about concerns of @jonsmirl and @chshu they mentioned above regarding testing?

jonsmirl, 2024-12-27:
I have not tried taking an internal bridge through testing

bilalmalik76, 2024-12-27:
@jonsmirl Thanks I will speak with ATL to discuss this.

closed 2025-01-23 by dhrishi