espressif/esp-matter #1199: From Where to find which cluster attributes are supported by various clusters in esp-matter

bilalmalik76, 2024-12-09:
From where to find which cluster attributes are supported by various clusters in esp-matter. I need to create PICS XML files for the followings manually because PICS generator does not generate these files: 

Access Control Enforcement Test Plan.xml
Group Communication.xml
Groups Cluster Test Plan.xml
Wi-Fi Network Diagnostics Cluster Test Plan.xml

jadhavrohit924, 2024-12-09:
@bilalmalik76 You can check the esp-matter [components](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp) cluster `create()` API.

VaishaliAvhale, 2024-12-09:
@bilalmalik76

Please find the attached updated PICS file and the steps to update it:
[PICS.zip](https://github.com/user-attachments/files/18061186/PICS.zip)

**Note:** In the attached PICS, I have selected all the mandatory items. If your application supports any optional features, please enable them as well.

**Steps to update PICS:**

Refer to the Core Matter Specification for core clusters and the Application Cluster Specification for application-specific clusters.
Open the PICS XML file using the PICS tool provided by CSA. Navigate to the specific cluster and select the mandatory items (optional items are not enabled in the Matter product but can be enabled by the end user) as per the PICS template. If any optional items are supported by your application, select them as well.
Validate the file and click the "Generate" button.

bilalmalik76, 2024-12-09:
@VaishaliAvhale The PICS files which you share do no have any attributes selected in any of these files:
Access Control Enforcement Test Plan.xml
Group Communication.xml
Groups Cluster Test Plan.xml
Wi-Fi Network Diagnostics Cluster Test Plan.xml

bilalmalik76, 2024-12-09:
In esp_matter_attribute.h for wifi_network_diagnotics, these attributed are being created. Does this mean all these attributes are supported? 

https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.h#L209

[image]

VaishaliAvhale, 2024-12-09:
It's my bad. Here are the updated PICS files:
[updated-PICS.zip](https://github.com/user-attachments/files/18064057/updated-PICS.zip)
>>In esp_matter_attribute.h for wifi_network_diagnotics, these attributed are being created. Does this mean all these attributes are supported?

Yes!

bilalmalik76, 2024-12-09:
@VaishaliAvhale how can I enabled this: 

> The Wi-Fi Network Diagnostics is an optional cluster, and optional things are not enabled in the Matter product by default. If you have enabled it, you can proceed with testing.

VaishaliAvhale, 2024-12-10:
>> How can I enable this?

Please refer to this link to add the Wi-Fi Network Diagnostics cluster:
[Adding Clusters - ESP Matter Documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#clusters).

>> From this file, it looks like core attributes are enabled for diagnostics_network_wifi (esp-matter/components/esp_matter/esp_matter_cluster.cpp)

Yes, the implementation is already present. You just need to call it.

bilalmalik76, 2024-12-10:
@VaishaliAvhale I think it is already being called when node config is passed to create node and it checks if it is wifi enabled device it adds diagnostics_network_wifi cluster also. 
[image]

VaishaliAvhale, 2024-12-10:
After the Node Configuration is passed, a Matter node is created, and the mandatory Root Node device type along with its mandatory clusters, attributes, events, and features are added on endpoint 0.

The diagnostics_network_wifi cluster is optional, and optional features are not enabled by default. Therefore, it needs to be added explicitly.

bilalmalik76, 2024-12-10:
Yes but I think line is adding it explicitly: 
```
#if CHIP_DEVICE_CONFIG_ENABLE_WIFI
    diagnostics_network_wifi::create(endpoint, &(config->diagnostics_network_wifi), CLUSTER_FLAG_SERVER);
#endif
```

bilalmalik76, 2024-12-19:
@VaishaliAvhale is this possible to get all PICS files for base endpoint0, on_off_plugin_unit and dimmable_plugin_unit end points. We tried generated with PICS tool and then tried to modify manually but still looks like some additional optional are enabled and some are missing, we did not enable any optional but used only those enabled by default in esp-matter.

bilalmalik76, 2024-12-19:
For example, we root node endpoint0, there is not group cluster. do we need group communication and group cluster test plan? 
[image]

bilalmalik76, 2024-12-19:
Following tests are failing, I am not sure if this is because of incorrect selection in PICS files:
[image]

bilalmalik76, 2024-12-20:
@VaishaliAvhale can you please provide base.xml file for matter over WiFi according to esp-matter default clusters and attributes which are enabled?

VaishaliAvhale, 2024-12-20:
Please find the attached base.xml
[Base.zip](https://github.com/user-attachments/files/18211776/Base.zip)

bilalmalik76, 2024-12-20:
@VaishaliAvhale I am getting this error with PICS tool:
[image]

bilalmalik76, 2024-12-20:
this does not seems to be a correct file with a lot items marked true for client:
[image]

VaishaliAvhale, 2025-01-02:
https://github.com/espressif/esp-matter/issues/1199#issuecomment-2557157043  - Click on enable author mode and try again

VaishaliAvhale, 2025-01-02:
It seems I made a mistake. This option should not have been set. 
Here is the correct base.xml. This is matter v1.3 base.xml extended color light pics For other matter version or other device type, the pics should be different
[Base.zip](https://github.com/user-attachments/files/18288762/Base.zip)

dhrishi, 2025-08-20:
Closing this now. Please re-open in case of any issues

closed 2025-08-20 by dhrishi