espressif/esp-matter #1200: Matter device authentication issue

tien1608, 2024-12-10:
I have a problem with device certification. When I DIY my ESP32C3 Mini device using Esp Zerocode, I get the problem: Device not certified. When I try Esp Launchpad, I don't get this problem. Why is that?
[image]

shubhamdp, 2024-12-11:
Matter has a certification program to certify a Matter product. These ESP Zerocode and ESP Matter launchpad binaries are non-certified evaluation firmwares. So, the error is expected, but you can still go ahead and add that device.

AFAIK, ESP Zerocode uses the custom non-certified custom certification declaration(CD) and hence you are seeing a warning.
For ESP Launchpad, answering with assumption that you used an example from esp-matter launchpad. It uses the test CD present in the SDK so may not show the warning assuming its a test device.

tien1608, 2024-12-11:
> Matter has a certification program to certify a Matter product. These ESP Zerocode and ESP Matter launchpad binaries are non-certified evaluation firmwares. So, the error is expected, but you can [...]

When I get the warning: Device not certified, but I try to add the device to Esp rainmaker it will give an error, adding to chip-tool also gets the same case. But when I use esp matter launchpad I don't get that warning, so I can add the device and use it normally.

BlackLordj, 2024-12-11:
> Matter có chương trình chứng nhận để chứng nhận sản phẩm Matter. Các tệp nhị phân ESP Zerocode và ESP Matter launchpad này là các chương trình cơ sở đánh giá chưa được chứng nhận. Vì vậy, lỗi là điều dễ hiểu, nhưng bạn vẫn có thể tiếp tục và thêm thiết bị đó.
> 
> AFAIK, ESP Zerocode sử dụng khai báo chứng nhận tùy chỉnh không được chứng nhận (CD) và do đó bạn thấy cảnh báo. Đối với ESP Launchpad, trả lời với giả định rằng bạn đã sử dụng ví dụ từ esp-matter launchpad. Nó sử dụng CD thử nghiệm có trong SDK nên có thể không hiển thị cảnh báo nếu cho rằng đó là thiết bị thử nghiệm.

I am also having a similar error like this but in my case on both espzero and launchpad it shows this warning and the device cannot be added to the matter network via the esp-rainmaker App

shubhamdp, 2024-12-11:
@BlackLordj if it's the same warning as mentioned, then you have two options right? "Still set up" and "Cancel". If you click still set up and device will be added to the ESP RainMaker app.

If you have different issue, please file a different issue, with device logs.

shubhamdp, 2024-12-11:
@tien1608 If you have problem adding the device to the rainmaker or chip-tool, please share the logs (paste logs in the file and attach the file) so that its easier to debug the problem that speculating what could be wrong.

BlackLordj, 2024-12-11:
> @BlackLordjnếu cảnh báo giống như đã đề cập, thì bạn có hai lựa chọn đúng không? "Vẫn thiết lập" và "Hủy". Nếu bạn nhấp vào vẫn thiết lập và thiết bị sẽ được thêm vào ứng dụng ESP RainMaker.
> 
> Nếu bạn gặp sự cố khác, vui lòng gửi sự cố khác kèm theo nhật ký thiết bị.

[image]
 an example here i am trying to build a project on esplaunchpad with esp32c6 then i chose still add device

BlackLordj, 2024-12-11:
> @BlackLordjnếu cảnh báo giống như đã đề cập, thì bạn có hai lựa chọn đúng không? "Vẫn thiết lập" và "Hủy". Nếu bạn nhấp vào vẫn thiết lập và thiết bị sẽ được thêm vào ứng dụng ESP RainMaker.
> 
> Nếu bạn gặp sự cố khác, vui lòng gửi sự cố khác kèm theo nhật ký thiết bị.

[image]
> I (2465) main_task: Returned from app_main()
I (12055) chip[DL]: BLE GAP connection established (con 0)
I (12055) chip[DL]: CHIPoBLE advertising stopped
I (13015) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13025) chip[BLE]: local and remote recv window sizes = 5
I (13035) chip[BLE]: selected BTP version 4
I (13035) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (13195) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 1
I (13195) chip[DL]: CHIPoBLE subscribe received
I (13205) NimBLE: GATT procedure initiated: indicate; 
I (13215) NimBLE: att_handle=18

I (13255) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13265) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13275) chip[EM]: >>> [E:28408r S:0 M:219881441] (U) Msg RX from 0:DE74EC8CF5E892B1 [0000] to 0000000000000000 --- Type 0000:20 (SecureChannel:PBKDFParamRequest) (B:68)
I (13285) chip[EM]: <<< [E:28408r S:0 M:250751929] (U) Msg TX from 0000000000000000 to 0:DE74EC8CF5E892B1 [0000] [BLE] --- Type 0000:21 (SecureChannel:PBKDFParamResponse) (B:154)
I (13295) NimBLE: GATT procedure initiated: indicate; 
I (13315) NimBLE: att_handle=18

I (13315) chip[SVR]: Commissioning session establishment step started
I (13345) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13435) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13445) chip[EM]: >>> [E:28408r S:0 M:219881442] (U) Msg RX from 0:DE74EC8CF5E892B1 [0000] to 0000000000000000 --- Type 0000:22 (SecureChannel:PASE_Pake1) (B:92)
I (13515) chip[EM]: <<< [E:28408r S:0 M:250751930] (U) Msg TX from 0000000000000000 to 0:DE74EC8CF5E892B1 [0000] [BLE] --- Type 0000:23 (SecureChannel:PASE_Pake2) (B:127)
I (13525) NimBLE: GATT procedure initiated: indicate; 
I (13525) NimBLE: att_handle=18

I (13585) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13615) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13625) chip[EM]: >>> [E:28408r S:0 M:219881443] (U) Msg RX from 0:DE74EC8CF5E892B1 [0000] to 0000000000000000 --- Type 0000:24 (SecureChannel:PASE_Pake3) (B:59)
I (13635) chip[EM]: <<< [E:28408r S:0 M:250751931] (U) Msg TX from 0000000000000000 to 0:DE74EC8CF5E892B1 [0000] [BLE] --- Type 0000:40 (SecureChannel:StatusReport) (B:30)
I (13655) NimBLE: GATT procedure initiated: indicate; 
I (13655) NimBLE: att_handle=18

I (13655) chip[SC]: SecureSession[0x40825048, LSID:56686]: State change 'kEstablishing' --> 'kActive'
I (13675) chip[SVR]: Commissioning completed session establishment step
I (13685) chip[DIS]: Updating services using commissioning mode 0
I (13685) chip[DIS]: CHIP minimal mDNS started advertising.
I (13695) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=0 cp=0
I (13705) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 65C2EFF6A380D7C5.
I (13715) chip[DIS]: mDNS service published: _matterc._udp
I (13715) chip[SVR]: Device completed Rendezvous process
I (13735) app_main: Commissioning session started
I (13735) app_main: Commissioning window closed
I (13705) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13745) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13755) chip[EM]: >>> [E:28409r S:56686 M:146688923] (S) Msg RX from 0:FFFFFFFB00000000 [0000] to 0000000000000000 --- Type 0001:02 (IM:ReadRequest) (B:51)
I (13775) chip[EM]: <<< [E:28409r S:56686 M:179581108] (S) Msg TX from 0000000000000000 to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:05 (IM:ReportData) (B:68)
I (13785) NimBLE: GATT procedure initiated: indicate; 
I (13795) NimBLE: att_handle=18

I (13825) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13835) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13845) chip[EM]: >>> [E:28410r S:56686 M:146688924] (S) Msg RX from 0:FFFFFFFB00000000 [0000] to 0000000000000000 --- Type 0001:02 (IM:ReadRequest) (B:51)
I (13855) chip[EM]: <<< [E:28410r S:56686 M:179581109] (S) Msg TX from 0000000000000000 to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:05 (IM:ReportData) (B:86)
I (13875) NimBLE: GATT procedure initiated: indicate; 
I (13875) NimBLE: att_handle=18

I (13915) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (13925) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (13935) chip[EM]: >>> [E:28411r S:56686 M:146688925] (S) Msg RX from 0:FFFFFFFB00000000 [0000] to 0000000000000000 --- Type 0001:02 (IM:ReadRequest) (B:52)
I (13945) chip[EM]: <<< [E:28411r S:56686 M:179581110] (S) Msg TX from 0000000000000000 to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:05 (IM:ReportData) (B:67)
I (13965) NimBLE: GATT procedure initiated: indicate; 
I (13965) NimBLE: att_handle=18

[... 559 lines cut ...]
W (236465) wifi:Haven't to connect to a suitable AP now!
[... cut ...]
I (265985) chip[DL]: bleAdv Timeout : Start slow advertisement
I (265985) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (265995) chip[DL]: Device already advertising, stop active advertisement and restart
I (265995) NimBLE: GAP procedure initiated: stop advertising.

I (266005) NimBLE: GAP procedure initiated: advertise; 
I (266015) NimBLE: disc_mode=2
I (266025) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (266035) NimBLE:

shubhamdp, 2024-12-11:
I gave a shot to this, used ESP32-C6, and ESP RainMaker android application (version 3.5.2 - 8b52741, region: Global). And was able to add device.

@BlackLordj  Can you confirm the rainmaker app and version that you used.

tien1608, 2024-12-12:
> @tien1608 If you have problem adding the device to the rainmaker or chip-tool, please share the logs (paste logs in the file and attach the file) so that its easier to debug the problem that speculating what could be wrong.
I tried to connect the esp32c3-mini device using Esp Zerocode to chip-tool but could not connect, I uploaded the logs file.
[connection.txt](https://github.com/user-attachments/files/18109887/connection.txt)
[consoledevice.txt](https://github.com/user-attachments/files/18109888/consoledevice.txt)

shubhamdp, 2024-12-19:
@BlackLordj @tien1608 do you have an update on this problem?

tien1608, 2024-12-19:
> @BlackLordj @tien1608 do you have an update on this problem?

@shubhamdp Thank you, the problem is due to the esp rainmaker version. I tried Android 3.5.0 version and didn't have this problem. When using version 3.5.1 it only happens.

closed 2024-12-19 by tien1608