espressif/esp-matter #1201: Matter device discovery issue

tien1608, 2024-12-10:
I encountered a problem in Light example. When I connect the device to Esp Rainmaker, I can't connect it to chip-tool at the same time. Even though I enabled advertising BLE on the device, chip-tool still can't detect it. The device I used is esp32c6.
[image]
[image]

[image]

shubhamdp, 2024-12-11:
I'd suggest you to please take a look at this simplified blog post which talks about Matter multi admin  feature https://developer.espressif.com/blog/matter-multi-admin-identifiers-and-fabrics/

Also, please check this small video about how multi admin feature works - https://youtu.be/-NGXwGz8EnQ

`matter ble adv start` won't move the device into commissioning state. Assuming you have added a device to first ecosystem, you have to open the commissioning window using the first ecosystem and then you can add the device to another ecosystem. For the second time, device will not advertise using BLE but the mdns.

I'd suggest you start with chip-tool first and then add the device to chip-tool as described in the [Commissioning and Control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) section. Then open the commissioning window using below command
```
chip-tool pairing open-commissioning-window 0x7283 1 180 1000 3840

# please check the command help for what value is what
chip-tool pairing open-commissioning-window node-id option window-timeout iteration discriminator
```

Then in the chip-tool logs you will see a either QR Code text or manual pairing code, then use that with the  ESP-RainMaker application to add that device to RainMaker ecosystem.

NOTE: While attaching logs, please paste the logs to the file and attach that file. Its very inconvenient to scrub through the such picturesque logs.

shubhamdp, 2024-12-11:
@tien1608 more detailed chip-tool guide can be found here: https://github.com/project-chip/connectedhomeip/blob/master/docs/development_controllers/chip-tool/chip_tool_guide.md

Please check the multi admin scenario section: https://github.com/project-chip/connectedhomeip/blob/master/docs/development_controllers/chip-tool/chip_tool_guide.md#multi-admin-scenario

shubhamdp, 2024-12-19:
@tien1608 Is this still a problem, else can you please close this issue.

tien1608, 2024-12-19:
> @tien1608 Is this still a problem, else can you please close this issue.

Thank you @shubhamdp , I did it your way and the problem is solved.

closed 2024-12-19 by tien1608