espressif/esp-matter #1203: Error while installing esp-matter on macOS

Corle, 2024-12-11:
**Describe the bug**
I'm following this guide: https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html
On step 2.2.1.1 Getting the Repository -> For Mac OS-X host
I'm running `./install.sh`
And getting these errors:
```
ERROR at /Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_build/facade.gni:187:7: Assertion failed.
      assert(_dep_is_in_link_dependencies,
      ^-----
/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_log:impl must be listed in the pw_build_LINK_DEPS build arg when the /Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_log:pw_log facade is in use. Please update your toolchain configuration.
See /Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_log/BUILD.gn:47:1: whence it was called.
pw_facade("pw_log") {
^--------------------
See //src/lib/support/BUILD.gn:355:5: which caused the file to be included.
    "$dir_pw_log:impl",
    ^-----------------
['gn', 'gen', '/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 402, in install_packages
    subprocess.check_call(
    ~~~~~~~~~~~~~~~~~~~~~^
        gn_cmd,
        ^^^^^^^
    ...<2 lines>...
        stderr=outs,
        ^^^^^^^^^^^^
    )
    ^
  File "/usr/local/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/subprocess.py", line 419, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 1002, in <module>
    sys.exit(main())
             ~~~~^^
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 994, in main
    return EnvSetup(**vars(parse())).setup()
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 579, in setup
    result = step(spin)
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 773, in virtualenv
    if not virtualenv_setup.install(
           ~~~~~~~~~~~~~~~~~~~~~~~~^
        project_root=self._project_root,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<17 lines>...
        use_pinned_pip_packages=self._use_pinned_pip_packages,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 410, in install_packages
    raise subprocess.CalledProcessError(
        err.returncode, err.cmd, ins.read()
    )
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/Users/corle/Developer/embedded/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/89874899809ed4b705184958b9d1cc14b92a4ed7
- ESP-IDF Commit Id: https://github.com/espressif/esp-idf/commit/a2f420a36ecf8cd19b443adde1d75e5eaecc6309

- Host Machine OS: macOS Sequoia 15.1.1
- Host Machine Python version: 3.13.0

shubhamdp, 2024-12-12:
@Corle Can you please re-try the installation process

```
rm -r connectedhomeip/connectedhomeip/.environment

source /path/to/esp-idf/export.sh
cd esp-matter
./install.sh
```

If this fails, please try this:
```
cd connectedhomeip/connectedhomeip
rm -rf .environment
source scripts/bootstrap.sh

cd -
source /path/to/esp-idf/export.sh
./install.sh --no-bootstrap
```

I could re-produce this once, but on retry I could not.

closed 2024-12-12 by Corle

marvinoc, 2025-01-19:
I'm having a similar problem, but I haven't been able to resolve it yet. I'm using Ubuntu 22.04 on a virtual machine.

"fatal: cannot change to 'describe': No such file or directory
Installing requirements from requirements.txt"