espressif/esp-matter #1205: Error on commissioning step 'AttestationVerification': 'Error CHIP:0x000000AC'

stefano055415, 2024-12-13:
Hi, I'm trying to commission a matter device with the controller example.

I tried first using the light example as a device and everything works fine.

I then tried to commission a certified device and I get the error Error on commissioning step 'AttestationVerification': 'Error CHIP:0x000000AC'

I understood that the problem is the test certificate that I am using in the controller, but I wanted to ask if it was possible to bypass this check and still conclude the commissioning at least for this first test phase

shripad621git, 2024-12-17:
@stefano055415 , in case you want to bypass the attestation verification, you can use `--bypass-attestation-verifier` in chip-tool which intends to continue commissioning in case of attestation failure.
The command `chip-tool pairing ble-wifi --help `will provide you a description of the option as shown below:
` --bypass-attestation-verifier:  Bypass the attestation verifier. If not provided or false, the attestation verifier is not bypassed. If true, the commissioning will continue in case of attestation verification failure.
`

stefano055415, 2024-12-17:
@shripad621git 

ok with chip-tool I can use the bypass, but with the example of the controller how do I do it?

The device is certified but the controller is not

shripad621git, 2024-12-18:
@stefano055415 , can you please follow the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#attestation-trust-storage) using the `Attestation Trust Store - Spiffs` and copying the 
production certs from `${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip/credentials/production/paa-root-certs/*.der` to paa_cert directory.
We will implement the main net DCL PAA cert fetching logic in the controller.

shripad621git, 2024-12-29:
@stefano055415 , any updates on the issue?

dhrishi, 2025-06-09:
>We will implement the main net DCL PAA cert fetching logic in the controller.

This was addressed sometime back with commit https://github.com/espressif/esp-matter/commit/75f52fa11817c40d3c7fcd92ae9e48635ce0e60b

closed 2025-06-09 by dhrishi