espressif/esp-matter #1206: Fish shell support for install snd export

PankajRedekar, 2024-12-15:
**Describe the question/query that you have**
esp-idf has support for fish shell but esp-matter not

**Additional context**
...

pimpalemahesh, 2024-12-18:
@PankajRedekar, please create the export.fish file with the below code in esp-matter directory. We will soon add support for the Fish shell in ESP-Matter.
```
function realpath_int
    set wdir (pwd)
    if [ "$wdir" = "/" ]
        set wdir ""
    end
    set arg $argv[1]

    switch "$arg"
        case /* 
            set scriptdir "$arg"
        case '*'
            set scriptdir "$wdir/(string replace -r '^\.\/' '' $arg)"
    end
    set scriptdir (string trim (dirname $scriptdir))
    echo $scriptdir
end

function esp_matter_export_main
    if not set -q ESP_MATTER_PATH
        # ESP_MATTER_PATH not set in the environment.
        # If using Fish, try to guess ESP_MATTER_PATH from script location.
        set self_path ""

        if test -n "$FISH_VERSION"
            set script_path (status --current-filename)
            set self_path (realpath $script_path)
        else
            echo "Could not detect ESP_MATTER_PATH. Please set it before sourcing this script:"
            echo "  set ESP_MATTER_PATH (add path here)"
            return 1
        end

        # Resolve the absolute path
        if string match -r "darwin*" "$OSTYPE" 
            set script_dir (realpath_int $self_path)
            set script_dir (cd $script_dir; pwd)
        else
            set script_name (readlink -f $self_path)
            set script_dir (dirname $script_name)
        end
        set -g -x ESP_MATTER_PATH $script_dir
        echo "Setting ESP_MATTER_PATH to '$ESP_MATTER_PATH'"
    end

    # PATH for gn
    set PATH $PATH $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/

    # PATH for host tools
    set PATH $PATH $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/out/host

    # export zap-cli path
    set ZAP_INSTALL_PATH $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/.environment/cipd/packages/zap
end

esp_matter_export_main

```

pimpalemahesh, 2025-01-10:
@PankajRedekar Adding support for different shells might cause internal errors. Please close the issue if it has been resolved.

dhrishi, 2025-01-25:
Closing this. Please reopen, with details, if you still face issues

closed 2025-01-25 by dhrishi