espressif/esp-matter #1207: BLE doesn't work after commissioning has ended

AlexVaq, 2024-12-15:
I'm writing a sleepy thread device that, after properly commissioned, connects to a BLE device with a predefined address to get some readings.

For some reason, the connection always fails, i.e., it stays in BLE_HS_EALREADY forever and never finishes. If a timeout is specified, the connection always results in time out.

I also tried a scan before connecting, but the scan returns events of zero length.

I'm using esp-idf 5.4.0 in vscode, and esp-matter is installed over the esp-idf extension. The board is an esp32c6.

The code is the icd_app example, plus at the end of app_main I add:

```
   /* Matter start */
    err = esp_matter::start(appEventCb);
    ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to start Matter, err:%d", err));
   /* ------- NEW CODE ------- */
   /* Check until we are ready to connect */
    do {if (bleReady && serverReady && commissioningReady) {readyToConnect = true;}; vTaskDelay(5000/portTICK_PERIOD_MS);} while(!readyToConnect);

  /* bleReady is true by default; becomes false if the event kCommissioningWindowOpened is received; becomes true if  the event kCHIPoBLEConnectionClosed is received and the number of fabrics is > 0
      commissioningReady is true by default; becomes false if the event kCommissioningWindowOpened is received; becomes true if the event kCommissioningComplete is received and the number of fabrics is > 0
      serverReady is false by default; becomes true if chip::DeviceLayer::DeviceEventType::kServerReady and the number of fabrics is > 0
      The number of fabrics comes from chip::Server::GetInstance().GetFabricTable().FabricCount() */

    /* Give some gracious time to the system to finish the set up, not sure if it's necessary, but removing it gave me problems sometimes */
    vTaskDelay(waitTime/portTICK_PERIOD_MS);

    /* Try to connect or scan */
    ESP_LOGI(TAG, "BLE Init");
    ble_init();
}

```
ble_init() could be a simple connection code:

```
        rc = ble_hs_id_infer_auto(0, &own_addr_type);
        if (rc != 0) {
            ESP_LOGE(tag, "Error determining address type. No valid address assigned to host");
            bleError = 128;
            return bleError;
        }
        rc = ble_gap_connect(own_addr_type, &addr, BLE_HS_FOREVER, NULL, ble_airwave_gap_event, NULL);

        if ((rc != 0) && (rc != BLE_HS_EALREADY) && (rc != BLE_HS_EDONE)) {
            ESP_LOGI(tag, "Error: Failed to connect to address %s\tCode %d", addr_str(addr.val), rc);
            bleError = 96;
            return rc;
        } else {
            ESP_LOGI(tag, "Current connection status %d", rc);
        }

```

or a simple scan code, very much like what @jonsmirl suggested in issue #906.

The connect code either times out or does not work, because a subsequent call to ble_gap_conn_find_by_addr fails.
The scan code returns 0 length events, which is absurd.

I have also tried calling ble_gatts_reset() and ble_svc_gap_init() before scanning or connecting, as it was suggested in the aforementioned issue. It didn't make any difference.

I have no idea what is preventing BLE from properly functioning after commissioning. Help would be very much appreciated.

shubhamdp, 2024-12-16:
Can you check if you have enabled  `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` option enabled. (ref: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-9-why-does-free-ram-increase-after-first-commissioning`)

AlexVaq, 2024-12-16:
Thank you for the answer. It is NOT enabled. I'm aware the BLE stack deinitializes when the option is enable, and can't be reinitialized again.

shubhamdp, 2024-12-17:
@AlexVaq Thanks for the confirmation that BLE is not de-initialized after commissioning is complete.

#### Peripheral use case
I cut-pasted the  from the [bleprph_advertise()](https://github.com/espressif/esp-idf/blob/v5.0.1/examples/bluetooth/nimble/bleprph/main/main.c#L143) from idf's `examples/bluetooth/nimble/bleprph` example, and I was able to advertise the service and can connect to it using nrf-connect app.

#### Central use case
For scanning purpose, cut-pasted code from [blecent_scan()](https://github.com/espressif/esp-idf/blob/v5.0.1/examples/bluetooth/nimble/blecent/main/main.c#L228) idf's `examples/bluetooth/nimble/blecent` example, and I was able to see the beacon advertised by nrf-connect app.

I'm attaching the [scan-or-adv-after-commissioning.patch](https://github.com/user-attachments/files/18162164/scan-or-adv-after-commissioning.patch) for reference, PTAL! You will have to modify the stuff in this to suit your requirements.

AlexVaq, 2024-12-17:
Thanks for the help.

Did you check that this works after matter commissioning? I will try to copy/paste your code and adapt it accordingly, but I'm foreseeing it won't fix the problem, since the code you are attaching is really, really similar to the one I'm already using.

To be clear, the problem is not with BLE per se; I tested my just-BLE code previously to make sure I could scan, connect to the device and read the data. The problem is that, for some reason, after matter is done with BLE, it seems I can't use BLE properly, and the code that works standalone won't work after initializing the matter server. But I'll try nonetheless and report back.

shubhamdp, 2024-12-17:
Yes, it works as well after the commissioning complete and on the subsequent boot when the device is already commissioned. I have attached the code snippet for the reference only (no need to start building from there). I know, even I have gone through the tutorials from those examples in order to build this one.

Agreed, It would just work, out of the box. Can you please share the sdkconfig, just curious to see if something in there making any difference.

AlexVaq, 2024-12-17:
I'm making progress thanks to your help. I found a mistake in my code that dismissed the device I was scanning, and instead reported zero-length fields. Now I can scan and find the device I wanted to connect. I noticed that **ble_gap_security_initiate** fails because it seems encryption is disabled at compile time, which might be the reason why I couldn't connect before.

Is this a matter feature, or is it possible to enable it?
Is it possible to enable the extended advertise feature?

Apart from these questions, I think the issue can be closed, since whatever problems I come across now will be different.
Thank you for your help, I really appreciate it.

shubhamdp, 2024-12-18:
> Is this a matter feature, or is it possible to enable it?

Matter does not use encryption from the bluetooth stack and relies on the packet encryptions specified in the Matter Specifications. I think you can do that by setting [characteristics flags](https://github.com/espressif/esp-nimble/blob/46f1139a055246d05c0744c2195c5f82155b5e6d/nimble/host/include/host/ble_gatt.h#L125) to support encrypted read/write or as per your needs.

> Is it possible to enable the extended advertise feature?

Yes, it is supported. In Matter its under [CONFIG_ ENABLE_EXTENDED_DISCOVERY](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L391) and for nimble please check [BT_NIMBLE_EXT_ADV](https://github.com/espressif/esp-idf/blob/cb3ac7429c73b6e7c6d79ae05de480d4b6706770/components/bt/host/nimble/Kconfig.in#L584)
NOTE: For Matter we have it implemented it as per Matter Specification Section 5.4.2.5.3.

I cannot close the issue. If you feel your questions are answered please do it yourself.

closed 2024-12-18 by AlexVaq