espressif/esp-matter #1208: How to commissioning matter over thread with esp32c6

giangnv1993, 2024-12-15:
**Describe the question/query that you have**
Currently, I'm working on matter and thread with esp32.
I setup include:
1. Ubuntu OS + chip tool installed on that
2. one esp32c6 and flash light_switch example from esp-matter

for commissioning, I used this command below, but always failed
`pairing ble-thread 2 hex:0e08000000000001000000030000184a0300001235060004001fffe002083cceaf03f46b3aa80708fd333f0818f26dd1051008c9fd20fff08fa3ccec18e9b0c8ded3030f4f70656e5468726561642d3763386101027c8a0410c220888959e93f545053ee18019991100c0402a0f7f8 20202021 3840 `

Do you have any link or any hints to support me in this case?
THanks so much!

shubhamdp, 2024-12-16:
@giangnv1993 can you please share the build steps that you have used to build the example. I'm assuming you are following steps from [developing the sdk](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html) section.

Please share the device logs, sdkconfig file, and commissioner logs.

If you have just used `idf.py set-target esp32c6` and `idf.py build` then it builds the example with Wi-Fi only configuration. You will need to configure that example to build with thread configuration.  Can you please try with the c6_thread configuration from light example.
```
idf.py -DSDKCONFIG_DEFAULTS=../light/sdkconfig.defaults.c6_thread set-target esp32c6 build
```

Dumb question but, asking this for re-assurance, do you have a Thread border router in the network?

giangnv1993, 2024-12-16:
> @giangnv1993 can you please share the build steps that you have used to build the example. I'm assuming you are following steps from [developing the [...]

Hi. Thanks for your support!
1. the system that I setup includes: 

- environment: ubuntu OS
- need: matter over a thread by chip tool
- what I did: built chip tool in Ubuntu, build esp-matter/example/light_swith with command only is idf.py set-target esp32c6` and `idf.py build. Yes, let me try with your recommendation, thanks

2. Yes, I have an OT_BR (esp32H2 (rcp) + esp32 (host/br),) and built successfully. 
what is the mandatory commissioning, Do you only need a chip tool with light_switch (esp32c6)?

shubhamdp, 2024-12-17:
@giangnv1993 In case of thread end devices, we need: 1. an end device (C6 in your case), 2. Border router (You are using the H2+ESP32 as OTBR), 3. Chip-tool.

Please let us know the update. Also, please share the chip-tool and ESP32-C6 logs next time (in the file) for further debugging help.

giangnv1993, 2024-12-17:
> @giangnv1993 In case of thread end devices, we need: 1. an end device (C6 in your case), 2. Border router (You are using the H2+ESP32 as OTBR), 3. Chip-tool.
> 
> Please let us know the update. Also, please share the chip-tool and ESP32-C6 logs next time (in the file) for further debugging help.

Hi @shubhamdp 
1. Yes. This is the log that I already commissioned and completed via chip-tool.
[Chip-tool-log.txt](https://github.com/user-attachments/files/18164441/Chip-tool-log.txt)
this file is a mess, activities are
- I performed commissioned with light_swith
- try to commission with other devices but failed

2. could you support me as a beginner?
- I built chip-tool completed in Ubuntu OS
- built light_Swith completed --> commission completed with chip tool via thread
- built OTBR, but I don't know what to do with this OTBR. tried to commission with chip-tool but failed
- I expect to build a system that includes light thread, switch thread, OTBR, and switch_light can control the light via matter fabric

shubhamdp, 2024-12-19:
@giangnv1993 Please see if you followed these actions...

1. Setup OTBR -- you can find more details about setting it up in [esp-thread-br](https://github.com/espressif/esp-thread-br) repo and in [basic OTBR example](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router#openthread-border-router-example).
2. Get the TBR dataset -- On the OTBR, try command `dataset active -x`. It will give you a long hex string which you will use during commissioning.
3. Build and flash the light_switch or any other example with thread configurations. Right now, very few examples has the default sdkconfig for C6-thread. So you will need to build with that. I had suggested a way earlier and can be re-purposed for other examples as well if they do not have C6-thread config file -- `idf.py -DSDKCONFIG_DEFAULTS=../light/sdkconfig.defaults.c6_thread set-target esp32c6 build`
4. Build the chip-tool, if you are using esp-matter and ran the `./install.sh` then its path is exported so no additional steps are needed.
5. At this point, make sure, OTBR and host running chip-tool is connected to the same wifi network. Now, commission the device using chip-tool.
```
chip-tool pairing ble-thread 2 hex:<data-set-retrieved-in-step-2> 20202021 3840
```

NOTE: Please make sure to use a random node-id (2 in above command). If you re-use the node-id it may not be able to find the device on operational network.

giangnv1993, 2024-12-19:
> @giangnv1993 Please see if you followed these actions...
>
> 1. Setup OTBR -- you can find more details about setting it up in [esp-thread-br](https://github.com/espressif/esp-thread-br) repo and in [...]

Hi @shubhamdp , thanks for your guide detail.

`chip-tool pairing ble-thread 2 hex:<data-set-retrieved-in-step-2> 20202021 3840`
This command will help thread device join matter Fabric? is that correct?
1. in case 'yes', then I don't know how chip-tool can control light thread by matter, because the example only is openthread, not matter implementation.
2. in case 'no', how can chip-tool control light via matter?

shubhamdp, 2024-12-19:
Yes, the command add the Matter thread-end device to the Matter fabric, it performs the necessary steps, to verify the device attestation, providing the operational data set to the thread-end device, and then making sure device is reachable on thread network.

Please check [commissioning and control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#commissioning-and-control) section from the documentation guide for how to send commands, read/write attributes, subscribe to events.

giangnv1993, 2024-12-19:
> Yes, the command add the Matter thread-end device to the Matter fabric, it performs the necessary steps, to verify the device attestation, providing the operational data set to the thread-end [...]

I'm building the thread matter light_switch in `esp-matter/example/light_switch`, is this correct for my expectation is "can be control light_switch use matter via thread"?

shubhamdp, 2024-12-19:
Yes, it does work with Matter via thread.

For starters, please check these blogs
 - https://developer.espressif.com/blog/what-does-matter-mean-to-you/
 - https://developer.espressif.com/blog/matter-thread-border-router-in-matter/
 
and if you are interested we have the complete series here: https://developer.espressif.com/blog/matter/

giangnv1993, 2024-12-19:
> Yes, it does work with Matter via thread.
>
> For starters, please check these blogs
>
> * https://developer.espressif.com/blog/what-does-matter-mean-to-you/
> * [...]

Hi @shubhamdp , thanks for your support!
Currently, I already built successes a system that includes: chip-tool, otbr and light example thread + matter.
But I'm facing 2 problems:
1. when commission light completed, I used command onoff on 2 1 to control light device, see the log seems to me that it control successfully, but I didn't see the led is on in light device. 
please find the log in below:
[chip-tool-log.txt](https://github.com/user-attachments/files/18201631/chip-tool-log.txt)
[matter_thread_device_light.txt](https://github.com/user-attachments/files/18201632/matter_thread_device_light.txt)
[OTBR_log.txt](https://github.com/user-attachments/files/18201633/OTBR_log.txt)

2. Still this system, I tried to add one more thread matter device that is `light_switch` in `esp-matter/example`
The steps are:
commissioning for light_switch first --> completed
next, do commissioning for light --> failed.
Here is the log:
[chip_tool.txt](https://github.com/user-attachments/files/18201669/chip_tool.txt)
[light_example_matter_thread.txt](https://github.com/user-attachments/files/18201670/light_example_matter_thread.txt)
[light_switch_matter_thread.txt](https://github.com/user-attachments/files/18201671/light_switch_matter_thread.txt)
[ot_br.txt](https://github.com/user-attachments/files/18201672/ot_br.txt)

Please give me some insight with 2 points above. Thanks so much!

goossensbas, 2024-12-21:
Hello,
I just updated my esp-matter repository, and i'm running into the same build problems.

When i build the example with the command "idf.py -DSDKCONFIG_DEFAULTS=../light/sdkconfig.defaults.c6_thread set-target esp32c6 build" the following error appears:
ManagerImpl.cpp:91:51: error: static assertion failed: Wi-Fi network endpoint id and Thread network endpoint id should not be the same.
I already did idf.py fullclean, but it didn't help.

shubhamdp, 2024-12-24:
@giangnv1993 
1. Yes, control is successful, but may be something went wrong with the led indicator driver and it was unable to turn on/off led.
```
W (913043) led_indicator: LED indicator does not have the hal_indicator_set_brightness function
```
Can you share the esp-idf and esp-matter commit id's used for this.

2. When commissioning light, I see that it failed just after requesting the PAI certificate, may be give it another try and see what happens?

shubhamdp, 2024-12-24:
@goossensbas which example are you building? Can you please provide more details:

- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):

goossensbas, 2024-12-24:
> @goossensbas which example are you building? Can you please provide more details:
> 
>     * ESP-Matter Commit Id:
> 
>     * ESP-IDF Commit Id:
> 
>     * SoC (eg: ESP32 or ESP32-C3):
> 
>     * Device Logs (Please attach the log file):

I think i found the problem. I used the sdkconfig for the C3 instead of the C6.
It compiles and i can commission it with CHIP :-)

giangnv1993, 2024-12-25:
> @giangnv1993
>
> 1. Yes, control is successful, but may be something went wrong with the led indicator driver and it was unable to turn on/off led.
>
> ```
> W (913043) led_indicator: LED indicator does not [...]

Hi, I used this command is possible turn on.off with light_example
`levelcontrol move-to-level 10 0 0 0 0x7283 0x1`
`levelcontrol move-to-level 100 0 0 0 0x7283 0x1`
> 
> 2. When commissioning light, I see that it failed just after requesting the PAI certificate, may be give it another try and see what happens?
Yes, I tried many times. 
Actually, I only commissioning a device, the second device is can't
Do you have any idea?

giangnv1993, 2024-12-26:
@shubhamdp Hi supporter, I built a light_sample app ( this one already worked with matter over wifi). Now, I want to thread over matter, and here is the system that I setup
1. chiptool on ubuntu serve (raspi 4 host)
2. esp32c6 for light_sample (already set target c6 and c6_thread as your guided)
3. otbr (esp32H2 + esp32)

the commissioning already completed, but I can't control on/off from chip tool. please see the logs in below:
[chip_tool.txt](https://github.com/user-attachments/files/18251953/chip_tool.txt)
[light_matter.txt](https://github.com/user-attachments/files/18251954/light_matter.txt)
[ot_br.txt](https://github.com/user-attachments/files/18251955/ot_br.txt)

shubhamdp, 2024-12-31:
```
E (623549) OPENTHREAD: Failed to get LL address, error: Invalid If index
```

This could be because, backbone netif is not set for the thread border router, and can be done by calling esp_openthread_set_backbone_netif() to set the backbone netif to the STA netif.
But, logs shows you calling `wifi connect -s -p`, and that does set the [backbone netif](https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_wifi_cmd.c#L298) for the tbr.

Can you please check the value of `CONFIG_LWIP_IPV6_NUM_ADDRESSES ` (or share the sdkconfig)  in thread-border-router app. In recent changes, it was bumped from 8 to 12 in the thread border router lib. In tbr example, this has to be exactly same as the value used in tbr lib. There is also a [check](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread_lwip_netif.c#L28) for that.

Can you share the logs from a single run with commit ids for esp-idf/esp-matter/esp-thread-br so that we can try to reproduce this. Also, if you do run this again, please share the logs from the single run execution where it failed. It's hard to map the logs from three different files.

shubhamdp, 2025-01-10:
@giangnv1993 any updates on this one?

dhrishi, 2025-01-21:
@giangnv1993 Please provide the requested details or close the issue if resolved